PX4/PX4-Autopilot #13893: The arming switch which is integrated in CUAV uavcan GPS can't work with UAVCAN

kevinuav, 2020-01-09:
Everything works perfectly,except the arming switch,magnetometer(rm3100), air pressure meter(ms5611)and the beeper. They can't work.

I think that it should be the driver issue.Could you help me to fix the problem?Thank you!

Kevin from CUAV

jinger26, 2020-01-09:
This is for the upcoming new GPS developed by CUAV. Called NeoV2

[image]

jinger26, 2020-01-09:
@kevinuav  do you have more technical details to share here?

kevinuav, 2020-01-10:
> @kevinuav do you have more technical details to share here?

[image]

Processor | STM32F417
Sensor |  
Compass | RM3100
Barometer | MS5611
GNSS |  
GNSS receiver | UBLOX Eighth Generation Navigation and Positioning Receiver (M8N)
GNSS concurrently | GPS/QZSS L1 C/A,GLONASS L10F ,BeiDou B1I,Galileo E1B/C,SBAS L1 C/A: WAAS, EGNOS, MSAS, GAGAN
Nav. update rate | 10HZ(MAX)
Nav. update rate | 2M
Acquisition | Cold start 26s ；Hot start 1s；Aided start 2s
Sensitivity | Tracking and nav. -167dBM；Cold start Hot star -148dBM；Reacquisition -160dBM
Protocol | UAVCAN
Upgrade firmware | support
Support Flight control type | Ardupilot/PX4*
Interface Type | GHR-04V-S
Input voltage | 5V
Operating temperature | -10~70℃
Size | 60*60*16MM
Weight | 33g

moreba1, 2020-10-15:
does this problem solved with 1.11.1 firmware?

dagar, 2020-10-15:
> does this problem solved with 1.11.1 firmware?

No, the problem is the DSDL is missing as it was added directly in Ardupilot (https://github.com/ArduPilot/ardupilot/tree/master/libraries/AP_UAVCAN/dsdl/ardupilot/indication) rather than UAVCAN public types (https://github.com/UAVCAN/public_regulated_data_types/tree/legacy-v0/uavcan). For PX4 to directly consume DSDL from Ardupilot we'd need the licensing to be clarified, although it would be even better if these basic messages were added to UAVCAN instead.