PX4/PX4-Autopilot #13896: EKF GPS blending does not account for sensor position

CarlOlsson, 2020-01-09:
There is a [TODO comment](https://github.com/PX4/Firmware/blob/32a4e3a942533d9863a38dbba6062cfb51f13b06/src/modules/ekf2/ekf2_main.cpp#L2141-L2141) in the code so this is a know issue, but I thought it is still worth documenting it here.
Currently EKF2 does not support blending the GPS module position offset on the vehicle.
For the position measurements it is maybe not crucial but for the velocity measurements it can be critical during fast attitude changes. E.g. during a rolling maneuver on a fixed wing with one GPS module in each wing.

dagar, 2020-01-09:
I'm not sure why it wasn't added initially, but one thing we're missing is offset for GPS2. Ultimately I'd like to have sensor metadata available system wide for all sensor types (per sensor instance) and not as EKF2 parameters.

dagar, 2020-01-09:
@priseborough or @RomanBapst can you comment on the GPS blending TODO?

https://github.com/PX4/Firmware/blob/32a4e3a942533d9863a38dbba6062cfb51f13b06/src/modules/ekf2/ekf2_main.cpp#L2141-L2144

RomanBapst, 2020-01-15:
@dagar Not sure what to say about it. It should be done I guess.