PX4/PX4-Autopilot #13905: Missing aid mask in EKF2

shabhu18, 2020-01-10:
We use GitHub issues only to discuss PX4 bugs and new features. For
questions about using PX4 or related components, please use [PX4 Discuss](http://discuss.px4.io/).

Px4 version: 1.10
Bitmask: as in  developer guide 

    0: use GPS
    1: use optical flow
    2: inhibit IMU bias estimation
    3: vision position fusion
    4: vision yaw fusion
    5: multi-rotor drag fusion
    6: rotate external vision
    7: GPS yaw fusion
    8: vision velocity fusion

Bitmask displayed in QGC

    0: use GPS
    1: use optical flow
    2: inhibit IMU bias estimation
    3: vision position fusion
    4: vision yaw fusion
    5: multi-rotor drag fusion
    6: rotate external vision
    7: GPS yaw fusion
   

Missing Vision velocity fusion in QGC with version 1.10 . Am i missing something ?

Thanks!

hamishwillee, 2020-01-12:
I thought perhaps the docs might be wrong, but it looks like this is definitely in the code: https://github.com/PX4/Firmware/blob/v1.10.0-rc3/src/modules/ekf2/ekf2_params.c#L604

shabhu18, 2020-01-13:
Here are the screenshot of my QGC 
[image]
[image]

The list of ekf_aid_mask pram list are different . Is there any other way to enable it ??

hamishwillee, 2020-01-13:
@bkueng Any ideas. The param is in a PX4v1.10rc but not in Firmware,

bkueng, 2020-01-13:
You need to use QGC daily or upload the firmware via QGC to get the metadata.
Or you can use `Advanced settings` and directly set the value.

hamishwillee, 2020-01-14:
@bkueng That sounds wrong. I thought there was a new QGC supposed to accompany every release?

bkueng, 2020-01-14:
Yes there usually is.

hamishwillee, 2020-01-15:
OK. Then closing this as resolved.

closed 2020-01-15 by hamishwillee