PX4/PX4-Autopilot #13923: make fail because nuttx submodule

lfeng17, 2020-01-12:
If I use git submodule update --init --recursive, it reports:

fatal: reference is not a tree: 4c2da2935399ca17eae0c9ad4e8822cb8efc2b0d
Unable to checkout '4c2da2935399ca17eae0c9ad4e8822cb8efc2b0d' in submodule path 'platforms/nuttx/NuttX/nuttx'

dagar, 2020-01-12:
Can you provide more detail? Which version of PX4/Firmware?

xdwgood, 2020-01-13:
yes,I encountered the same problem. 
I encountered this error when I tried to compile the latest master branch:
```
fatal: reference is not a tree: 4c2da2935399ca17eae0c9ad4e8822cb8efc2b0d
Unable to checkout '4c2da2935399ca17eae0c9ad4e8822cb8efc2b0d' in submodule path 'platforms/nuttx/NuttX/nuttx'
fatal: reference is not a tree: 4c2da2935399ca17eae0c9ad4e8822cb8efc2b0d
Unable to checkout '4c2da2935399ca17eae0c9ad4e8822cb8efc2b0d' in submodule path 'platforms/nuttx/NuttX/nuttx'

```
Latest master branch

xdwgood, 2020-01-13:
same as issue: https://github.com/PX4/NuttX/issues/80

xdwgood, 2020-01-13:
I try `git reset --hard  commit number` , Found that since this PR was merged, the problem occurred

https://github.com/PX4/Firmware/pull/13787

julianoes, 2020-01-15:
Fixed.

closed 2020-01-15 by julianoes