PX4/PX4-Autopilot #13928: the priority of rc controller or QGC for fixed wing

jinchengde, 2020-01-13:
recently, I talked with @Antiheavy for some feature I wish to add for fixed wing.

**1. abort landing if throttle position is larger than the threshold. #13842**

**2. reset mission after mission had finished #10288**

- mission have not landing waypoint, or user trigger abort landing, so the plane will loiter after mission had finished.
- switch the flight mode from MISSION to any others like STABLIZE, and switch back to MISSION.
- the mission will reset and plane will restart from the first waypoint.(current the plane will still loiter)

**3. abort landing and go to LAND_START_WAYPOINT to start approach again**

- user trigger abort landing
- reset mission and find the LAND_START_WAYPOINT, set the waypoint as next
- start approach again 

and you will find that both of these feature is trigger by the rc controller, and also could be trigger by QGC, but in my use case, rc controller have high priority than QGC, espically if your fixed wing is too large and fast(like gas engine but not foam), the pilot have the response for the safe, specially during the stage of TAKEOFF and LANDING, the QGC have large lag and only be used to monitor the status of vehicle

Antiheavy, 2020-01-13:
Hi @jinchengde   Is there a problem with priority?  It seems to me both QGC and these potential new features controlled by the RC controller have equal priority - and that is okay.

> **3. abort landing and go to LAND_START_WAYPOINT to start approach again**
> 
>     * user trigger abort landing
> 
>     * reset mission and find the LAND_START_WAYPOINT, set the waypoint as next
> 
>     * start approach again

I look forward to reviewing a PR for this one.  It should be optionally enabled by parameter in my opinion.

jinchengde, 2020-01-16:
> reset mission after mission had finished #10288
had rebase in #13948 

> 3. abort landing and go to LAND_START_WAYPOINT to start approach again

> I look forward to reviewing a PR for this one. It should be optionally enabled by parameter in my opinion.
had implement if #13948 is merged because if we trigger abort landing, we had already set the current mission index as land start waypoint
https://github.com/PX4/Firmware/blob/master/src/modules/navigator/mission.cpp#L1415-L1421
so just reset the mission(switch flight mode) and the vehicle will go to the land start waypoint
I had test it in SITL

Antiheavy, 2020-01-16:
> had implemented if #13948 is merged because if we trigger abort landing, we had already set the current mission index as land start waypoint in
> https://github.com/PX4/Firmware/blob/master/src/modules/navigator/mission.cpp#L1415-L1421
> so just reset the mission(switch flight mode) and the vehicle will go to the land start waypoint directly.
> I had test it in SITL.

I forgot about that.  Now I'm concerned about how the current abort landing behavior will be changed.  I'll comment in https://github.com/PX4/Firmware/pull/13948 too.