PX4/PX4-Autopilot #13937: Adding the availability of a command in docs

kvmanohar22, 2020-01-13:
I am going through the dev guide and specifically the section on [commands](https://dev.px4.io/master/en/middleware/modules_command.html). Since this is generated from source code, the latest commands are generated in the docs. Many of the commands (`listener`, `led_control` etc) are not available in the previous FMU versions. It would be helpful the reader to know from which FMU versions the corresponding command is introduced. 

Looking forward to hearing your thoughts on this.

hamishwillee, 2020-01-13:
IMO helpful but not enough to justify the effort. 

We re-release every 6 months or so and encourage people to update to the latest version. If you're on the latest version the fact that something wasn't present in the previous version is irrelevant. Of course not everyone updates, but the effort of "back porting" this information not worth it.

Also, at the point you add the code you don't know what version it is going into (though you have a "fair guess" in many cases.

@bkueng What do you think?

LorenzMeier, 2020-01-13:
It's a fair ask, but a typical one where I would say: We're happy to accept a contribution for a versioned generator (you could back-track via git hashes which files to build). But the maintenance team has a lot on its plate and is generally focused on looking forward, not backward. In the Linux world companies like Redhat maintain things forever, but the Linux Kernel project doesn't.

closed 2020-01-13 by LorenzMeier