PX4/PX4-Autopilot #13938: Bad altitude estimation with firmware update

60ericleal60, 2020-01-13:
I am preparing for flight with the master build forked from a week age and am getting poor altitude estimations I would expect if the vehicle lacked a range sensor. The correct distance is in the Mavlink inspector, range_aid is enabled and ekf2_hgt_mode is set to range sensor. I am using the lidar lite v3 in i2c mode.

With the same vehicle parameters in the 1.10.0 release build a good altitude estimation with the correct bottom clearance.

dagar, 2020-01-14:
Can you share a flight log? https://logs.px4.io/

60ericleal60, 2020-01-14:
https://logs.px4.io/plot_app?log=c857ee4a-9bbb-46af-b8af-e4d435781386

julianoes, 2020-02-14:
I don't see the distance listed in the log. Did you fly high enough for the distance to be measurable. Have you tried to check mavlink telemetry for the distance sensor?

60ericleal60, 2020-02-14:
The minimum height for the range sensor is set to 0.05m which is less than the height of the landing gear and yes the range output in the mavlink inspector appears to be correct. I am having the problem in 1.11 but not 1.10. For flight I have been modifying the firmware on the px4 flow board and pixhawk to fly very near the ground.

julianoes, 2020-02-14:
So even with 1.11/master you see the lidar information in MAVLink but not in the log?