PX4/PX4-Autopilot #13942: Unable to execute ORBIT mode from Mavros

mohit-ludhiyani, 2020-01-14:
Issue: I am trying to implement orbit mode using mavros.  During flight the drone switches mode to orbit mode. But, after reaching circle periphery the altitude starts decreasing. 

The ROS node publishes MAV_CMD_DO_ORBIT using rosservice "mavros/cmd/command". The command parameters sent are as follows:

` mavros_msgs::CommandLong orbit_cmd;
    orbit_cmd.request.broadcast = false;
    orbit_cmd.request.command = 34;  
    orbit_cmd.request.param1 = 6; // Radius of the circle
    orbit_cmd.request.param2 = 0.5; //Tangential Velocity.
    orbit_cmd.request.param3 = 0; //ORBIT_YAW_BEHAVIOUR 
    orbit_cmd.request.param5 = gps_home_pose.x(); //Center point latitude
    orbit_cmd.request.param6 = gps_home_pose.y(); //Center point longitude
    orbit_cmd.request.param7 = ground_altitude_amsl_mean + takeoff_height; //Center point altitude amsl`

In Qground control  mode switches from takeoff to orbit mode. But, once drone reaches the circle periphery it's altitude starts reducing. Also, the orbit execution status shows all the parameters correctly except the z component is always nan. 

The same code is working fine with JMAVSIM simulator. The z component in simulation shows the actual value published by ROS node. 

i am taking amsl value from /mavros/altitude.

dayjaby, 2020-01-14:
The orbit mode is semi-manual. The flight task reacts to throttle inputs by changing the altitude. Do you have any (virtual) rc switched on?

mohit-ludhiyani, 2020-01-15:
I issue a takeoff command through mavros. I just use rc for arming the drone after that everything is autonomous. Hence when flight mode changes to orbit the throttle value is 0. 
Is that causing problem? Is there any way that i can remove dependence from rc. 

My objective is to circle around point of interest

dayjaby, 2020-01-15:
It should be fine if you keep throttle at centre (0.5 throttle). Arent you using a rc that goes to throttle centre by default. I recommend this kind of setup, because if you have to takeover with POSITION mode, you already have a stable RC input.

With the other sticks you can change the orbit direction, change the radius and change the velocity during orbit by the way.

mohit-ludhiyani, 2020-01-15:
Hey, you were correct the rc throttle was causing the problem.  Thus this means i can only control altitude by rc in orbit mode.

But, i want to control the altitude of orbit without rc may be through onboard computer and keep rc for emergency. And, therefore i was trying to control altitude height using param7 in MAV_CMD_DO_ORBIT(altitude setpoint).

Is param7 of MAV_CMD_DO_ORBIT not used by px4 firmware for altitude adjustment?

mohit-ludhiyani, 2020-01-18:
Is it possible to control altitude in Orbit mode using MAV_CMD_DO_ORBIT param7??

Thanks in Advance!!

dayjaby, 2020-01-18:
Yes, you can send another CMD_DO_ORBIT with another altitude. The Orbit mode contains 2 stages: 
1. reaching the circle
2. orbitting

If you send another CMD_DO_ORBIT, then stage 1 will be performed again.

julianoes, 2020-01-24:
@MaEtUgR you might be able to help here.