PX4/PX4-Autopilot #13946: Firmware broken on Crazyflie 2.0 hardware -- unable to connect to the nsh console using mavlink

spheroidal, 2020-01-15:
**Describe the bug**
Unable to connect to the nsh console of crazyflie 2.0. I did a bisection on the Firmware commits and found that the connection problem started from [this merge](https://github.com/PX4/Firmware/pull/11256). I was able to compile and flash the codes, but I was unable to connect to the nsh console using mavlink afterwards. [A similar issue](https://github.com/PX4/Firmware/issues/11750) was reported before but was not resolved.

**To Reproduce**
Steps to reproduce the behavior:
1. Compile the default PX4 firmware for the crazyflie 2.0. 
2. Connect your computer to the vehicle via a usb port and flash the vehicle.
3. Try to connect to the nsh console using the command *./Tools/mavlink_shell.py /dev/ttyACM0* (You may use *dmesg* to see if the crazyflie 2.0 corresponds to /dev/ttyACM0 or some other path.)
4. The connection get stuck forever after displaying "Connecting to MAVLINK..." and no error message was displayed. 

**Expected behavior**
The connection succeeds and the console displays *nsh>*

dagar, 2020-01-24:
I'm ordering hardware to debug and ultimately add to the test rack.

dagar, 2020-07-27:
Should be fixed in https://github.com/PX4/Firmware/pull/15421.

closed 2021-01-17 by dagar