PX4/PX4-Autopilot #13947: make px4_sitl jmavsim can't work

liangpu597, 2020-01-15:
My System is : ubuntu 16.04+ PX4
I follow the instructions, my step is
1. wget https://raw.githubusercontent.com/PX4/Firmware/master/Tools/setup/ubuntu.sh
wget https://raw.githubusercontent.com/PX4/Firmware/master/Tools/setup/requirements.txt
2. source ubuntu.sh
3.  wget https://www.eprosima.com/index.php/component/ars/repository/eprosima-fast-rtps/eprosima-fast-rtps-1-7-1/eprosima_fastrtps-1-7-1-linux-tar-gz -O eprosima_fastrtps-1-7-1-linux.tar.gz
tar -xzf eprosima_fastrtps-1-7-1-linux.tar.gz eProsima_FastRTPS-1.7.1-Linux/
tar -xzf eprosima_fastrtps-1-7-1-linux.tar.gz requiredcomponents
tar -xzf requiredcomponents/eProsima_FastCDR-1.0.8-Linux.tar.gz
4. (cd eProsima_FastCDR-1.0.8-Linux && ./configure --libdir=/usr/lib && make -j2 && sudo make install)
(cd eProsima_FastRTPS-1.7.1-Linux && ./configure --libdir=/usr/lib && make -j2 && sudo make install)
rm -rf requiredcomponents eprosima_fastrtps-1-7-1-linux.tar.gz
5. git clone https://github.com/PX4/Firmware.git
6.   cd Firmware
git submodule update --init --recursive
7. make px4_sitl jmavsim

and the errors : 
`make px4_sitl jmavsim
[1/1] cd /home/tony/PX4/code/Firmware/...4/code/Firmware/build/px4_sitl_default
SITL ARGS
sitl_bin: /home/tony/PX4/code/Firmware/build/px4_sitl_default/bin/px4
debugger: none
program: jmavsim
model: none
src_path: /home/tony/PX4/code/Firmware
build_path: /home/tony/PX4/code/Firmware/build/px4_sitl_default
empty model, setting iris as default
SITL COMMAND: "/home/tony/PX4/code/Firmware/build/px4_sitl_default/bin/px4" "/home/tony/PX4/code/Firmware"/ROMFS/px4fmu_common -s etc/init.d-posix/rcS -t "/home/tony/PX4/code/Firmware"/test_data
INFO  [px4] Creating symlink /home/tony/PX4/code/Firmware/ROMFS/px4fmu_common -> /home/tony/PX4/code/Firmware/build/px4_sitl_default/tmp/rootfs/etc
0 WARNING: setRealtimeSched failed (not run as root?)

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [param] selected parameter default file eeprom/parameters_10016
[param] parameter file not found, creating eeprom/parameters_10016
  SYS_AUTOCONFIG: curr: 0 -> new: 1
* SYS_AUTOSTART: curr: 0 -> new: 10016
  BAT_N_CELLS: curr: 0 -> new: 3
  CAL_ACC0_ID: curr: 0 -> new: 1311244
  CAL_ACC_PRIME: curr: 0 -> new: 1311244
  CAL_GYRO0_ID: curr: 0 -> new: 2294028
  CAL_GYRO_PRIME: curr: 0 -> new: 2294028
  CAL_MAG0_ID: curr: 0 -> new: 197388
  CAL_MAG_PRIME: curr: 0 -> new: 197388
  CAL_BARO_PRIME: curr: 0 -> new: 6620172
  COM_DISARM_LAND: curr: 2.0000 -> new: 0.1000
  COM_OBL_ACT: curr: 0 -> new: 2
  COM_OF_LOSS_T: curr: 0.0000 -> new: 5.0000
  COM_RC_IN_MODE: curr: 0 -> new: 1
  EKF2_ANGERR_INIT: curr: 0.1000 -> new: 0.0100
  EKF2_GBIAS_INIT: curr: 0.1000 -> new: 0.0100
  COM_ARM_EKF_AB: curr: 0.0017 -> new: 0.0050
  EKF2_REQ_GPS_H: curr: 10.0000 -> new: 0.5000
  MC_PITCH_P: curr: 6.5000 -> new: 6.0000
  MC_PITCHRATE_P: curr: 0.1500 -> new: 0.2000
  MC_ROLL_P: curr: 6.5000 -> new: 6.0000
  MC_ROLLRATE_P: curr: 0.1500 -> new: 0.2000
  MPC_HOLD_MAX_Z: curr: 0.6000 -> new: 2.0000
  MPC_Z_VEL_I: curr: 0.1000 -> new: 0.1500
Buildfile: /home/tony/PX4/code/Firmware/Tools/jMAVSim/build.xml
  MPC_Z_VEL_P: curr: 0.2000 -> new: 0.6000
  MPC_XY_P: curr: 0.9500 -> new: 0.8000
  MPC_XY_VEL_P: curr: 0.0900 -> new: 0.2000
  MPC_XY_VEL_D: curr: 0.0100 -> new: 0.0160
  MPC_JERK_MIN: curr: 8.0000 -> new: 10.0000
  MPC_ACC_HOR_MAX: curr: 5.0000 -> new: 3.0000
  NAV_ACC_RAD: curr: 10.0000 -> new: 2.0000
  NAV_DLL_ACT: curr: 0 -> new: 2
  RTL_DESCEND_ALT: curr: 30.0000 -> new: 5.0000
  RTL_LAND_DELAY: curr: -1.0000 -> new: 5.0000
  RTL_RETURN_ALT: curr: 60.0000 -> new: 30.0000
  SDLOG_MODE: curr: 0 -> new: 1
  SDLOG_PROFILE: curr: 3 -> new: 131
  SDLOG_DIRS_MAX: curr: 0 -> new: 7
  SENS_BOARD_X_OFF: curr: 0.0000 -> new: 0.0000
  SENS_DPRES_OFF: curr: 0.0000 -> new: 0.0010
  TRIG_INTERFACE: curr: 4 -> new: 3
* RTL_DESCEND_ALT: curr: 5.0000 -> new: 10.0000
* RTL_LAND_DELAY: curr: 5.0000 -> new: 0.0000
  PWM_MAX: curr: 2000 -> new: 1950
  PWM_MIN: curr: 1000 -> new: 1075
* SYS_AUTOCONFIG: curr: 1 -> new: 0
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
INFO  [simulator] Waiting for simulator to connect on TCP port 4560

make_dirs:

compile:

create_run_jar:

copy_res:

BUILD SUCCESSFUL
Total time: 0 seconds
Options parsed, starting Sim.
Starting GUI...
3D [dev] 1.6.0-pre12-daily-experimental daily

com.jogamp.opengl.GLException: X11GLXDrawableFactory - Could not initialize shared resources for X11GraphicsDevice[type .x11, connection :0, unitID 0, handle 0x0, owner false, ResourceToolkitLock[obj 0x6c94ca2d, isOwner false, <173ee24, 1524a068>[count 0, qsz 0, owner <NULL>]]]
	at jogamp.opengl.x11.glx.X11GLXDrawableFactory$SharedResourceImplementation.createSharedResource(X11GLXDrawableFactory.java:326)
	at jogamp.opengl.SharedResourceRunner.run(SharedResourceRunner.java:297)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.NullPointerException
	at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:688)
	at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:580)
	at jogamp.opengl.x11.glx.X11GLXDrawableFactory$SharedResourceImplementation.createSharedResource(X11GLXDrawableFactory.java:297)
	... 2 more
com.jogamp.opengl.GLException: J3D-Renderer-1: createImpl ARB n/a but required, profile > GL2 requested (OpenGL >= 3.1). Requested: GLProfile[GL3bc/GL3bc.hw], current: 3.0 (Compat profile, compat[ES2], FBO, hardware) - 3.0 Mesa 18.0.5
	at jogamp.opengl.x11.glx.X11GLXContext.createImpl(X11GLXContext.java:418)
	at jogamp.opengl.GLContextImpl.makeCurrentWithinLock(GLContextImpl.java:759)
	at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:642)
	at jogamp.opengl.GLContextImpl.makeCurrent(GLContextImpl.java:580)
	at javax.media.j3d.JoglPipeline$QueryCanvas.doQuery(JoglPipeline.java:8615)
	at javax.media.j3d.JoglPipeline$QueryCanvas.access$100(JoglPipeline.java:8566)
	at javax.media.j3d.JoglPipeline.createQueryContext(JoglPipeline.java:6562)
	at javax.media.j3d.Canvas3D.createQueryContext(Canvas3D.java:4609)
	at javax.media.j3d.Canvas3D.createQueryContext(Canvas3D.java:3606)
	at javax.media.j3d.Renderer.doWork(Renderer.java:461)
	at javax.media.j3d.J3dThread.run(J3dThread.java:271)

DefaultRenderingErrorListener.errorOccurred:
CONTEXT_CREATION_ERROR: Renderer: Error creating Canvas3D graphics context for queryProperties()
graphicsDevice = X11GraphicsDevice[screen=0]
canvas = me.drton.jmavsim.Visualizer3D$CustomCanvas3D[canvas0,0,0,0x0,invalid]
Exception in thread "main" java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.eclipse.jdt.internal.jarinjarloader.JarRsrcLoader.main(JarRsrcLoader.java:61)
Caused by: java.lang.IllegalStateException: Canvas3D: Non-recoverable graphics configuration error
	at javax.media.j3d.Canvas3D.queryProperties(Canvas3D.java:3595)
	at me.drton.jmavsim.Visualizer3D.createEnvironment(Visualizer3D.java:269)
	at me.drton.jmavsim.Visualizer3D.<init>(Visualizer3D.java:215)
	at me.drton.jmavsim.Simulator.<init>(Simulator.java:189)
	at me.drton.jmavsim.Simulator.main(Simulator.java:939)
	... 5 more
X11Util.Display: Shutdown (JVM shutdown: true, open (no close attempt): 2/2, reusable (open, marked uncloseable): 0, pending (open in creation order): 2)
X11Util: Open X11 Display Connections: 2
X11Util: Open[0]: NamedX11Display[:0, 0x7f51c000c720, refCount 1, unCloseable false]
X11Util: Open[1]: NamedX11Display[:0, 0x7f51a0020c80, refCount 1, unCloseable false]
`

Please tell me how to solve ? I'am fresh man, thanks!

hamishwillee, 2020-01-15:
The opengl warnings are all driver issues. I get them when I run on a VM. Try HEADLESS - you won't see the while flying view, but you can still drive the vehicle on QGC.
```
HEADLESS=1 make px4_sitl jmavsim
```

liangpu597, 2020-01-15:
> The opengl warnings are all driver issues. I get them when I run on a VM. Try HEADLESS - you won't see the while flying view, but you can still drive the vehicle on QGC.
> 
> ```
> HEADLESS=1 make px4_sitl jmavsim
> ```
I use : HEADLESS=1 make px4_sitl jmavsim

and it still can't show jmavsim , the output is : 
`-- PX4 version: v1.11.0-beta1-241-gfb0bf26
-- PX4 config file: /home/tony/PX4/code/Firmware/boards/px4/sitl/default.cmake
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is GNU 5.4.0
-- The C compiler identification is GNU 5.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working CXX compiler using: Ninja
-- Check for working CXX compiler using: Ninja -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler using: Ninja
-- Check for working C compiler using: Ninja -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Found PythonInterp: /usr/bin/python (found version "2.7.12") 
-- Found PY_jinja2: /usr/lib/python2.7/dist-packages/jinja2  
-- build type is RelWithDebInfo
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-161-g94484f0
-- Configuring done
-- Generating done
-- Build files have been written to: /home/tony/PX4/code/Firmware/build/px4_sitl_default
[22/714] git submodule src/drivers/gps/devices
[29/714] git submodule src/lib/ecl
[30/714] git submodule mavlink/include/mavlink/v2.0
[31/714] git submodule Tools/jMAVSim
[714/714] cd /home/tony/PX4/code/Firmw...4/code/Firmware/build/px4_sitl_default
SITL ARGS
sitl_bin: /home/tony/PX4/code/Firmware/build/px4_sitl_default/bin/px4
debugger: none
program: jmavsim
model: none
src_path: /home/tony/PX4/code/Firmware
build_path: /home/tony/PX4/code/Firmware/build/px4_sitl_default
empty model, setting iris as default
SITL COMMAND: "/home/tony/PX4/code/Firmware/build/px4_sitl_default/bin/px4" "/home/tony/PX4/code/Firmware"/ROMFS/px4fmu_common -s etc/init.d-posix/rcS -t "/home/tony/PX4/code/Firmware"/test_data
INFO  [px4] Creating symlink /home/tony/PX4/code/Firmware/ROMFS/px4fmu_common -> /home/tony/PX4/code/Firmware/build/px4_sitl_default/tmp/rootfs/etc
0 WARNING: setRealtimeSched failed (not run as root?)

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [param] selected parameter default file eeprom/parameters_10016
[param] parameter file not found, creating eeprom/parameters_10016
  SYS_AUTOCONFIG: curr: 0 -> new: 1
* SYS_AUTOSTART: curr: 0 -> new: 10016
  BAT_N_CELLS: curr: 0 -> new: 3
[... 61 lines cut ...]
Options parsed, starting Sim.
GUI not enabled
INFO  [simulator] Simulator connected on TCP port 4560.
Init MAVLink
ERROR [commander] LED: open /dev/led0 failed
WARN  [commander] LED init failed
INFO  [init] Mixer: etc/mixers/quad_w.main.mix on /dev/pwm_output0
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 14570 remote port 14550
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
INFO  [logger] Opened full log file: ./log/2020-01-15/02_38_12.ulg
INFO  [mavlink] MAVLink only on localhost (set param MAV_BROADCAST = 1 to enable network)
INFO  [px4] Startup script returned successfully
pxh> INFO  [ekf2] Mag sensor ID changed to 197388
INFO  [ecl/EKF] 672000: EKF aligned, (baro height, IMU buf: 22, OBS buf: 14)
INFO  [ecl/EKF] 672000: reset position to last known position
INFO  [ecl/EKF] 672000: reset velocity to zero
INFO  [ecl/EKF] 9888000: GPS checks passed (WGS-84 origin set)
INFO  [ecl/EKF] 14416000: reset position to GPS
INFO  [ecl/EKF] 14416000: reset velocity to GPS
INFO  [ecl/EKF] 14416000: commencing GPS fusion
`

### How to do next?  please help me!

hamishwillee, 2020-01-15:
It looks like it built and is starting to fuse GPS. What do you see on QGroundControl? What is the problem?
@julianoes I'm end of day later - can you look at this when you get in to see I'm not missing anything obvious?

julianoes, 2020-01-15:
I think all is fine. You just don't see the jMAVSim application because you have started it `HEADLESS`.

Also: please when pasting code use ``` around it, thanks.

liangpu597, 2020-01-15:
@ My first step is : make px4_sitl jmavsim
and it's error is the topest in my write
and @hamishwillee told me use: HEADLESS=1 make px4_sitl jmavsim
and the erroris also exits.

Please tell me how to do ?

jinchengde, 2020-01-16:
It looks like some issue running on virtual machine, I also meet that before

liangpu597, 2020-01-17:
@jinchengde 
Yes , I run in vmware . and i don't how to solve it.

julianoes, 2020-01-18:
What error? Everything seems fine. The window does not appear with `HEADLESS=1`

closed 2020-04-18 by julianoes

iQyan, 2020-06-30:
Solution: 
> MESA_GL_VERSION_OVERRIDE=3.0 make px4_sitl jamvsim