PX4/PX4-Autopilot #13954: Angular rate setpoints not working as it used to

okalachev, 2020-01-15:
Making automated flips on PX4 is possible with angular velocity setpoints in OFFBOARD mode. However, after commit https://github.com/PX4/Firmware/commit/8bcf54688a1cfe7a042a1df847c120d73fcc0264 the behaviour has regressed.

Before, it was possible to make PX4 do a diagonal flip (by pitch and roll simultaneously). After, it only rotates by pitch, ignoring the roll setpoint. It reproduces in both SITL and a real drone.

**To Reproduce**
Send setpoints:
pitch rate = 30 radians
roll rate = 30 radians
thrust = 0.2

**Expected behavior**
Diagonal flipping (as it was in v1.8.2).

**Actual behavior**
Pitch flipping.

@bkueng, please, help!

bkueng, 2020-01-16:
Can you provide a log?

okalachev, 2020-01-16:
Diagonal flip attempts in SITL:

* Before the commit (working): https://logs.px4.io/plot_app?log=18cdb6a5-6288-4f94-ac38-5c9e3919646d
* After the commit (not working): https://logs.px4.io/plot_app?log=1fae65a8-cfad-430d-84d8-81469e9748f7

bkueng, 2020-01-17:
Can you enable `MC_AIRMODE`? Then it will work.

The problem is that the demands you send are extremely high, and w/o airmode, roll demands have priority over pitch at the lower end.

okalachev, 2020-01-17:
Thanks for the answers, @bkueng.

But honestly, I don't see any sensible reason to prioritize roll demands over pitch in quadcopters, even if airmode is off. Besides, in previous version such setpoints worked correctly.

Now, it turns out, that with airmode off (which is the default), the user doesn't have a way to make the drone rotate in pitch and roll axes at maximum speed. They either should somehow know the maximum possible rotation speed of the drone, either get the drone rotating over some axes faster than over another.

Finally, the commit description reads "The behavior for non-airmode and roll/pitch airmode should be the same as before". But the behavior is not the same as before.

bresch, 2020-01-20:
Hi @okalachev , I'm sorry that default behavior (airmode disabled) changed for your acrobatic flights but this mode hasn't been designed for this application. This is exactly for this kind of application that airmode have been created and performs really well (please try it! :) ).

The reason it worked before was more a side effect than a feature and obtaining the same behavior now might not be straightforward: the current strategy is to desaturate one axis at a time and having the same roll and pitch priority would require a 2D desaturation.

To conclude, if you can derive and/or implement a nice solution to make an efficient 2D desaturation, I'll be more than happy to review it (improvements are always welcome!), but my general answer is : if you want to perform acrobatic flights, enable airmode.