PX4/PX4-Autopilot #13956: Issue with landing PixRacer FMUV4_K66, FMUv5

Tony3dr, 2020-01-15:
**Describe the bug**
When flying the vehicle in Mission mode and trigger RTL, upon landing the vehicle does not descend vertically but drifts away with the wind. I had to take command of the vehicle in position mode and landed the vehicle manually. This is an issue that we have seen constantly, with the PixRacer and K66, FMUv5. **We are seeing this issue when the wind is 6 m/s and over.** 

**To Reproduce**
Steps to reproduce the behavior:
1. Drone switched on 
2. Uploaded mission to the vehicle
3. Took off in Mission Mode 
4. All commands were sent form QGC (Armed and takeoff)
5. Once all waypoint completed vehicle switched to RTL
6. When the vehicle was landing, it did not descend vertically, drifted away with the wind and landed the vehicle manually

**Expected behavior**
The vehicle should land and disarm.

**Log Files and Screenshots**
This is the latest PR tested with the PixRacer and issue persisted, Reference [#13678](https://github.com/PX4/Firmware/pull/13678)
Logs: https://review.px4.io/plot_app?log=67a2120c-a473-43a3-a556-6a542f62f7b1

Issue noticed on Reference  [#13701](https://github.com/PX4/Firmware/pull/13701
), wind 10.2 m/s, PixRacer.

Issue noticed on Reference [#13745](https://github.com/PX4/Firmware/pull/13745#event-2883456760), wind 10.2 m/s, PixRacer V4.
Log: https://review.px4.io/plot_app?log=719ab88b-d993-47c8-a9ca-eaf429d4577e

Issue noticed on Reference [#13613](https://github.com/PX4/Firmware/pull/13613#event-2837043850), 10.2 m / s, PixRacer V4.
Logs: https://review.px4.io/plot_app?log=b0f1f872-fc9e-41ee-89e3-4e146862efc7
Log: https://review.px4.io/plot_app?log=a8a1391c-06c0-475e-943a-48bea5da2e24

Issue noticed on Reference [#13065](https://github.com/PX4/Firmware/pull/13065), NXP_FMUK66_V3, 
Logs: https://review.px4.io/plot_app?log=792db7ac-8738-4bdd-8a03-aa6d88b371b3

**Drone (please complete the following information):**
**PixRacer**
Frame: DJI F450
Flight Controller: PixRacer 
GPS: Readytosky M8N GPS Module Built-in Compass
Motors: T Motor 2212-920kv
Propellers: DJI Phantom Built-in Nut Upgrade Propellers - MR PH 9.4x5 
Receiver: FrSky D4R-II 2.4G 4CH ACCST Telemetry Receiver
Telemetry:Holibro Telemetry, 100mW Telemetry Radio V3 915MHz
ESC: Hobbywing XRotor 20A APAC Brushless ESC
Battery: 3S 5200mAh(Hobbyking)

**NXP:**
Vehicle description:  RDDRONE-FMUK66
Flight Controller: FMUK66 
Frame: ReadytoSky LJI X4 500
Motors: ReadytoSky RS2212 920 kV CW & CCW  
Propellers: LJI 9450 self-locking propellers
GPS: Holybro Pixhawk 4 GPS
ESC: ReadytoSky 2-6S 40 A OPTO ESCs
Battery: 3S 5200mAh 

**Additional context**
We will test when windy on Master and Stable and see if the issue persists.

julianoes, 2020-01-16:
FYI @MaEtUgR

devararendy, 2020-02-08:
Woww I also have same problem.
Unfortunately I only have the video. I don't have the flight log.

I'm using PixRacer with PX4 v1.9.2 firmware. And I'm using ublox m8n + LNA. I got 18 satelites and the compass also working correctly.

And actually the battery is still enough, I only set the  minimum voltage per cell a bit higher.

When I started Auto Mode, it flew a couple meter then changed to Land Mode due to the drone detect low battery. But instead of landing, this drone flew away and landed quite far and I dont know where is it.

I only have the video from GCS.

A few second before loss os at 13:30

Here is the video.
[Raw footage PixRacer before flew away](https://youtu.be/zNJodNBDaaU)

julianoes, 2020-02-12:
@devararendy sorry about your fly away! What concerns me in the video is that you repeatedly get preflight errors about GPS velocity not being accurate enough. I'm not sure if that has something to do with it.

devararendy, 2020-02-13:
> @devararendy sorry about your fly away! What concerns me in the video is that you repeatedly get preflight errors about GPS velocity not being accurate enough. I'm not sure if that has something to do with it.

Thank You very much @julianoes hmm yeaaa, btw why i got GPS velocity not being accurate ? cuz I got 18 satellites.
And if you see the last mode is Land Mode, but it's not descending. is it due to the GPS velocity is not accurate, so the FC trying to maintain the horizontal velocity to zero before descending ? Hopefully you can help us figure it out. cuz i would like to build another drone again 

Thank You very much
Best Regards

devararendy

julianoes, 2020-02-14:
> btw why i got GPS velocity not being accurate ?

Yes, I wonder too. A log would probably tell us.

MaEtUgR, 2020-04-16:
Thanks to @Tony3dr for reporting this issue in this exemplary manner. I remembered it when debugging different issues today and found the cause:
[image] From 
> Issue noticed on Reference #13701, wind 10.2 m/s, PixRacer.

- Since #11913 RTL usually skips the descend state and directly goes to the land state
- RTL is designed to first quickly and with full horizontal control descend in the descend state and only switch to land state at the very end to do the final touch down on the ground. This works assuming the drone took off from the the home position and hence knows the relative altitude above the ground and hence when to start the final approach and probably breaks with rally points where the elevation is not exactly known.
- RTL in Navigator sets the mission item to `NAV_CMD_LAND` to indicate that it's just the final approach. https://github.com/PX4/Firmware/blob/8b76c0c0238a2142b7981d05c98c449f22a7f233/src/modules/navigator/rtl.cpp#L382
- That gets converted to a `position_setpoint_triplet.current.type` of `SETPOINT_TYPE_LAND` https://github.com/PX4/Firmware/blob/8b76c0c0238a2142b7981d05c98c449f22a7f233/src/modules/navigator/mission_block.cpp#L612
- The flight task generating the setpoints and constraints for that case limits the maximum tilt to `MPC_TILTMAX_LND` which is by default 12° https://github.com/PX4/Firmware/blob/8b76c0c0238a2142b7981d05c98c449f22a7f233/src/lib/flight_tasks/tasks/AutoMapper/FlightTaskAutoMapper.cpp#L137
- Depending on the vehicle and external conditions e.g. wind, VTOL the distrubance is high and the position controller is just not able anymore to allocate enough tilt
- The drone flies away with the wind as reported.

I'm looking into the proper way to fix this.

PR #14702 (2020-04-21): FlightTasks: do not adjust tilt limit of the position control
**Describe problem solved by this pull request**
Adjusting the tilt limit can lead to diverging position control and should only be used by setting a sanity limit for the controller and not to adjust during the descent phase of a Land or RTL. Otherwise it leads to flyaways in important failsafe modes when there's stronger disturbance e.g. wind.

**Describe your solution**
I removed all occurances where flight tasks sets a tilt limit. All of them except for the one in AutoMapper's `_prepareLandSetpoints()` were setting the maximum re [... cut]
closed 2020-04-21 by LorenzMeier