PX4/PX4-Autopilot #13961: Tolchaine_PX4 + Eclipse (Windows 10 x64) numpy error

astekachev, 2020-01-16:
Eclipse (Windows10 64) does not see Python numpy package
When compiling a project in bash, the package is recognized and compilation succeeds.
Log from bash
`$ make posix
-- PX4 version: v1.11.0-beta1-280-g8505dddba3
-- PX4 config file: /cygdrive/c/PX4/home/Firmware/boards/px4/sitl/default.cmake
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is GNU 7.4.0
-- The C compiler identification is GNU 7.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working CXX compiler: /usr/bin/c++.exe
-- Check for working CXX compiler: /usr/bin/c++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.9", minimum required is "3")
-- Found PY_jinja2: /usr/lib/python3.6/site-packages/jinja2
-- Found PY_toml: /usr/lib/python3.6/site-packages/toml
-- Found PY_numpy: /usr/lib/python3.6/site-packages/numpy
-- build type is RelWithDebInfo
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-161-g94484f0
-- Configuring done
-- Generating done
-- Build files have been written to: /cygdrive/c/PX4/home/Firmware/build/px4_sitl_default
[0/693] git submodule src/drivers/gps/devices
[9/693] git submodule src/lib/ecl
[183/693] git submodule mavlink/include/mavlink/v2.0
[693/693] Linking CXX shared library examples__dyn_hello.px4mod`

When I try to compile a project through Eclipse, the numpy package is not recognized and an error is thrown. Interestingly, other packages such as jinja and toml are recognized normally.
Eclipse Log
`make posix 
-- PX4 version: v1.11.0-beta1-280-g8505dddba3
-- PX4 config file: /cygdrive/c/PX4/home/Firmware/boards/px4/sitl/default.cmake
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is GNU 7.4.0
-- The C compiler identification is GNU 7.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working CXX compiler: /usr/bin/c++.exe
-- Check for working CXX compiler: /usr/bin/c++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.9", minimum required is "3") 
-- Found PY_jinja2: /usr/lib/python3.6/site-packages/jinja2  
-- Found PY_toml: /usr/lib/python3.6/site-packages/toml  
CMake Error at /usr/share/cmake-3.14.5/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  couldn't find python3 module numpy:

  		

  for debian systems try: 		

  	sudo apt-get install python3-numpy 		

  or for all other OSs/debian: 		

  	sudo -H pip3 install numpy

   (missing: PY_NUMPY) 
Call Stack (most recent call first):
  /usr/share/cmake-3.14.5/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  cmake/px4_find_python_module.cmake:67 (find_package_handle_standard_args)
  CMakeLists.txt:284 (px4_find_python_module)

-- Configuring incomplete, errors occurred!
See also "/cygdrive/c/PX4/home/Firmware/build/px4_sitl_default/CMakeFiles/CMakeOutput.log".
Error: /cygdrive/c/PX4/home/Firmware/build/px4_sitl_default is not a directory
make: *** [Makefile:195: px4_sitl_default] Error 1
"make posix" terminated with exit code 2. Build might be incomplete.`

I already tried to remove numpy and reinstall it, but when I try to install the numpy package again from bash, errors are output and it does not install.
Maybe someone has already encountered this problem and was able to solve it? I will be glad of any help.
Best regards

astekachev, 2020-01-19:
After some investigations, it appears: /usr/lib/lapack path has to be added to current $PATH
These actions solve this issue.

astekachev, 2020-01-19:
current $PATH:

> /usr/lib/lapack;${CYGWIN_HOME}/bin;C:/Program Files/Java/jdk-13.0.1/bin/server;C:/Program Files/Java/jdk-13.0.1/bin;C:/PX4/toolchain/cygwin64/bin;C:/PX4/toolchain/cygwin64/lib/python2.7/site-packages;C:/PX4/toolchain/cygwin64/lib/python3.6/site-packages;C:/PX4/toolchain/cygwin64/lib/python3.6/site-packages/numpy;C:/PX4/toolchain/gcc-arm/arm-none-eabi/bin;C:/PX4/toolchain/gcc-arm/bin;C:/PX4/toolchain/jdk/bin;C:/PX4/toolchain/apache-ant/bin;C:/PX4/toolchain/genromfs/;C:/PX4/toolchain/misc/;C:/PX4

astekachev, 2020-01-19:
Please add the following lines to setup-environment.bat x
`REM backup windows path variable to detect windows git installation
SET WINDOWS_PATH=%PATH%

REM path to Cygwin Unix Environment
SET PATH=%PX4_DIR%\toolchain\cygwin64\bin
REM path to python library
SET PATH=%PATH%;%PX4_DIR%\toolchain\cygwin64\lib
REM path to python numpy lib
SET PATH=%PATH%;%PX4_DIR%\toolchain\cygwin64\lib\lapack
REM path to GCC for ARM-NONE-EABI
SET PATH=%PATH%;%PX4_DIR%\toolchain\gcc-arm\arm-none-eabi\bin
REM path to GCC for ARM Compiler
SET PATH=%PATH%;%PX4_DIR%\toolchain\gcc-arm\bin
REM path to Java Developement Kit
SET PATH=%PATH%;%PX4_DIR%\toolchain\jdk\bin
REM path to Apache Ant, a Java build tool
SET PATH=%PATH%;%PX4_DIR%\toolchain\apache-ant\bin
REM path to genromfs
SET PATH=%PATH%;%PX4_DIR%\toolchain\genromfs\
REM path to miscellaneous executables e.g. custom shell scripts
SET PATH=%PATH%;%PX4_DIR%\toolchain\misc\`

astekachev, 2020-01-19:
Please add the following lines to install-cygwin-python-packages.bat:

```
REM install needed Python 2 packages under cygwin
CALL bash -c "pip2 --disable-pip-version-check install --user toml pyserial pyulog empy pyyaml cerberus pyros-genmsg"
REM install needed Python 3 packages under cygwin
CALL bash -c "pip3 --disable-pip-version-check install --user toml pyserial pyulog empy pyyaml cerberus pyros-genmsg"
```

MaEtUgR, 2020-01-21:
@astekachev Thanks for the report and the additional insights! Be aware that the scripts of the Cygwin toolchain are all versioned and the installer built by CI here: https://github.com/PX4/windows-toolchain
Could you please add your suggestions as pull-requests with a description of what exactly they are doing? That would help a lot.

I didn't really understand what was the issue in your use case. I'm guessing that a login shell commanded with `bash --login` could potentially have solved your problem. I did that for the CI call here: https://github.com/PX4/Firmware/blob/635ceccfdb9cf6bdd56ccfed5acb66d947959d50/appveyor.yml#L38

astekachev, 2020-01-21:
@MaEtUgR Thank you for your interest! 
The main problem was in Eclipse CDT environment that did not see the python numpy and toml package. In "run-console.bat" environment everything works fine. 
After some research, it turned out that:
1. To correctly determine Python toml package in the Eclipse CDT environment, you need to pip install with the --user attribute.
2. Eclipse CDT must have the address to \lib\lapack in the %PATH to correctly determine Python numpy package.

MaEtUgR, 2020-04-30:
That means if I'd switch to the `--user` python module installation when setting up the toolchain here: https://github.com/PX4/windows-toolchain/blob/ce3bac26991f7317fee74c33135e32855a2636cb/toolchain/cygwin64/install-cygwin-python-packages.bat#L12 it might solve your issue and the CI and `run-console.bat` requiring to wrap all build commands in a login shell with `bash --login` or `bash -l`. Let me see if that's easy to check.

MaEtUgR, 2020-04-30:
No I checked it doesn't work because numpy and jinja2 need to be installed through the packet manager (see https://github.com/PX4/windows-toolchain/blob/ce3bac26991f7317fee74c33135e32855a2636cb/toolchain/cygwin64/install-cygwin-px4.bat#L16-L17) because they contain binaries that we don't want to build from source (out of the box pip install fails) and the packet manager installs them in a system wide location.

So the official solution to have them recoginzed the right way is to use a login shell. So make your automatic build command e.g. `bash --login -c "cd Firmware && make px4_sitl"`.

closed 2020-04-30 by MaEtUgR