PX4/PX4-Autopilot #13968: EKF2 timestamp issue

CarlOlsson, 2020-01-17:
Already known issue (mentioned [here](https://github.com/PX4/Firmware/pull/11396)) so this issue is just for documentation.

EKF2 uses the sensor_combined.timestamp field everywhere instead of system time.

https://github.com/PX4/Firmware/blob/75c8fb12e48058966fff59795d0d11020604ca41/src/modules/ekf2/ekf2_main.cpp#L777-L777

This means e.g. that the fields published by ekf2 have the wrong timestamp.

It can also lead to other issues, e.g. we had this line of code to check if the airspeed data is not old
```
if (now - _airspeed.timestamp < 1_s)
```

However, this sometimes evaluated to true because _airspeed.timestamp is newer than now and then the subtraction returned a negative value

dagar, 2020-01-17:
We should fix the usage in EKF2 immediately. The broader issue that I've wanted to tackle for quite some time now is that timestamps need to be uORB metadata and not used for other purposes when convenient.

It's not quite ready yet, but I'll open an initial PR that moves setting the timestamp into uORB::Publication to solicit feedback.

In cases where we specifically want a sample timestamp I've started added that as a separate field. https://github.com/PX4/Firmware/pull/13955

dagar, 2020-01-17:
https://github.com/PX4/Firmware/pull/13970

dagar, 2020-07-19:
We need to review the remaining cases.