PX4/PX4-Autopilot #13980: camera feedback documentation

moreba1, 2020-01-19:
Hi

There is not a documentation for camera feedback setting up , wiring and testing between camera and pixhawk autopilot pins .

https://github.com/mavlink/qgroundcontrol/issues/8216#issuecomment-575943525

moreba1, 2020-01-26:
@Antiheavy 
Hi
do you know camera feedback setup ?

moreba1, 2020-01-26:
https://github.com/PX4/Firmware/issues/8247
https://discuss.px4.io/t/camera-triggering-feedback/7764/5
https://discuss.px4.io/t/camera-feedback-wiring-and-setting/13569/5

hamishwillee, 2020-01-28:
Camera triggering is documented in https://docs.px4.io/master/en/peripherals/camera.html . Hotshoe support has not been added (I believe - @DanielePettenuzzo can you please confirm?)

moreba1, 2020-02-12:
@AuterionWrikeBot  @DanielePettenuzzo

hamishwillee, 2020-02-13:
Yes, confirmed - hotshoe support not yet present. @DanielePettenuzzo will get it in at some point.

moreba1, 2020-05-13:
@hamishwillee 
Can anyone else help?

hamishwillee, 2020-05-18:
Nope. Essentially this feature does not exist, which sux because much of the work was done and has been allowed to lapse. 
So if you need really need this you might have to implement it yourself.

hamishwillee, 2020-08-16:
This is going to be partially covered by this update: https://github.com/PX4/px4_user_guide/pull/781

Essentially prior to FMUv5 the camera capture feedback is provided via Other FMU PWM pin 6 (hardcoded) for camera capture. FMUv5 uses a hard coded point. The camera_trigger mavlink message gets emitted when the capture is triggered AND on capture.

For MAVLink cameras obviously things aren't wired up at all, but the camera emits a message when it captures.

moreba1, 2020-08-24:
"Some cameras can provide image capture signaling to a flight controller camera-capture pin (this enables more accurate image timestamp than the time that the trigger command is sent)."

What is the requirement of signal of camera to capture pin ? can we connect the hot-shoe signal of the camera directly?

hamishwillee, 2020-08-25:
@bkueng ^^^ https://github.com/PX4/Firmware/issues/13980#issuecomment-679241698

bkueng, 2020-08-25:
> can we connect the hot-shoe signal of the camera directly?

Yes I think so. I've never done this myself though.

moreba1, 2020-08-26:
> 
> 
> > can we connect the hot-shoe signal of the camera directly?
> 
> Yes I think so. I've never done this myself though.

Where is related code? I will see that how signal input defined? (voltage pulse change value )

hamishwillee, 2020-09-28:
I've updated the docs in https://github.com/PX4/px4_user_guide/pull/807

The simple answer is that you cannot just connect a hotshoe direct to the  camera capture pin because the voltage levels are not likely to be compatible. There are however hotshoe adapters that you can use.

I believe the info is now correct though I have a final request for confirmation of the voltage level the pin expects: https://github.com/PX4/px4_user_guide/pull/807#issuecomment-700316410

closed 2020-09-28 by hamishwillee