PX4/PX4-Autopilot #13997: How to yaw Standard VTOL without rudder in configuration?

109nik, 2020-01-21:
**Describe the bug**
I am making a new air frame in 'VTOL configuration without rudder' and I desire to achieve yaw with ailerons and elevator only and not using rudder . Air frame has two ailerons, elevator and pusher in fixed wing mode and 4 motors in multi copter mode. I need help in mixing actuators in VTOL configuration without rudder. I can see Plane A-tail and Plane V-tail without rudders. 

- Is it possible to achieve same feature in VTOL configuration ?
- Is there any sample mixer file for 'VTOL without rudder' configuration?

moreba1, 2020-01-24:
4 motors dont work in fixedwing modes . they are for takeoff and landing .

sfuhrer, 2020-01-29:
@109nik did I understand you correctly, you want to use the multicopter motors to control yawing in fixed-wing flight? This would need much more changes than only in the mixer. Like @moreba1 pointed out, the multicopter motors are only active in hover mode, and not in fixed-wing mode.

sfuhrer, 2020-01-29:
@109nik while certainly improving flight performance, active yaw control is not crucially required for most fixed-wing vehicles. If your system doesn't fly nice without a rudder I would recommend to either add one or change the aerodynamic properties in a way that it becomes passively stable in yaw (the easiest thing there is to add/increase the vertical stabilizer).