PX4/PX4-Autopilot #14027: RADIO_STATUS handling assumes a certain radio type

LorenzMeier, 2020-01-26:
There are several assumptions built into the RADIO_STATUS packet handling that need to be changed:

  - If the message is ever received the link is set to a 3DR radio, which is by now an outdated assumption that will have side effects: https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_main.cpp#L1554
  - It enables software flow control, without checking if the buffer field is set - a non-3DR radio might have set this to an invalid value (we might need an "unknown" value for this field): https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_main.cpp#L1558
  - If radio status times out, the link gets reset to generic - this is probably harmless, but shouldn't happen (if there was a RADIO_STATUS, then it still is the detected type): https://github.com/PX4/Firmware/blob/master/src/modules/mavlink/mavlink_main.cpp#L1533

MaEtUgR, 2020-01-27:
Seems like it broke in https://github.com/PX4/Firmware/commit/930ac8d4fe032e533e2499b15fe6efe75ca6d63f#diff-a067fe2f10623e822f3af3bfd7b74f00R1614-L1617
@dagar I presume it should still be dependent on if you are using a 3DR radio and not set it right?

dagar, 2020-01-27:
> Seems like it broke in [930ac8d#diff-a067fe2f10623e822f3af3bfd7b74f00R1614-L1617](https://github.com/PX4/Firmware/commit/930ac8d4fe032e533e2499b15fe6efe75ca6d63f#diff-a067fe2f10623e822f3af3bfd7b74f00R1614-L1617)
> @dagar I presume it should still be dependent on if you are using a 3DR radio and not set it right?

Probably not, we have no real way to detect if it's a "3DR" (aka SiK) radio or not. The idea that receiving RADIO_STATUS means 3DR radio though can definitely die. Then the next question is how we should be handling RADIO_STATUS (https://mavlink.io/en/messages/common.html#RADIO_STATUS) in general. Whenever I've taken a close look at what's actually happening in the mavlink module I've found the crude flow control via RADIO_STATUS (without CTS/RTS connected) actually didn't work well at all. 

I'll continue in the PR (#14032).

MaEtUgR, 2020-01-27:
> I've found the crude flow control via RADIO_STATUS (without CTS/RTS connected) actually didn't work well at all

I would have expected that from reading the code...

Cool, thank you!

PR #14032 (2020-02-03): Only do SW flow control for 3DR radio
Since the removal of the `type` field in  https://github.com/PX4/Firmware/pull/10395, the type of a radio was always assumed to be 3DR. Removal made sense, because the corresponding mavlink message does not contain a type field.

Consequently, when other components report a `radio_status`, the mavlink instance would be automatically using software flow control. This isn't always desired.

This fixes #14027
closed 2020-02-03 by dagar