PX4/PX4-Autopilot #14029: Make documentation more precise - FMU position

dimon777, 2020-01-27:
https://dev.px4.io/v1.9.0/en/advanced/parameter_reference.html

EKF2_IMU_POS_[X|Y|Z] parameters need better clarification:
[image]

What is: "position of IMU in body frame"? Relative to what: CoG, Geometric Center, something else?

PX4 team - **please set simple standard: precision in aviation is paramount**. Thanks.

bresch, 2020-01-27:
Hi @dimon777 , I agree it's not explicitly written, but since any object rotates about its CG, it's quite usual to have the origin of the body frame there.
Could you please suggest a modification [here](https://github.com/PX4/Firmware/blob/15835ce7a3b3213df4a83a6a292a07ddf3de79c2/src/modules/ekf2/ekf2_params.c#L821-L846) to help the next one that will have the same question or tell me which description would you like to see for those parameters please?

Refs:
https://www.grc.nasa.gov/www/k-12/airplane/rotations.html
https://en.wikipedia.org/wiki/Flight_dynamics_(fixed-wing_aircraft)#Reference_frames

hamishwillee, 2020-01-28:
How about adding "(relative to vehicle centre of gravity)" to each of those. E.g.

> X position of IMU in body frame *(relative to vehicle centre of gravity)*.

@dimon777 If you agree, can you create a PR with the modification ([here](https://github.com/PX4/Firmware/blob/15835ce7a3b3213df4a83a6a292a07ddf3de79c2/src/modules/ekf2/ekf2_params.c#L821-L846) as @bresch suggested)?

hamishwillee, 2020-01-28:
And @bresch - can you confirm what you'd think best for above (in case @dimon777 does not respond)?

bresch, 2020-01-29:
Thanks @hamishwillee for jumping in. I think the clearest would be:
```
X position of IMU in body frame (forward axis with origin relative to vehicle centre of gravity).
```
And replace "forward" by "right" and "down" for the Y and Z axes, respectively. (Forward-Right-Down (FRD) is IMO simpler to understand and technically correct)

hamishwillee, 2020-01-30:
THanks @bresch - FRD is indeed a much clearer way of stating this. PR created.

PR #14060 (2020-01-30): EKF params - clarify body frame origin/orientation
Fixes #14029
closed 2020-01-30 by mhkabir