PX4/PX4-Autopilot #14053: NXP_FMUK66_v3 - VTOL - Airspeed sensor not recogniced

dk7xe, 2020-01-29:
**Describe the bug**
Holybro PX4 Airspeed sensor not recogniced on NXP_FMUK66_v3 with current master / stable release.
Airframe: Caipiroshka Duo Tailsitter

**To Reproduce**
Steps to reproduce the behavior:
1. Drone switched on with Airspeed sensor connected to I2C/NFC port
2. check Sensor tab in QGC. No airspeed sensor
3. start mavlik console and enter 'ms5611 start'
4. message: 'WARN [ms5611] no device on bus x' (2 lines; one for bus 1 another bus 2)

**Expected behavior**
Airspeed sensor is recogniced by PX4 sw

**Log Files and Screenshots**
*Always* provide a link to the flight log file:
Log of v1.10.1 stable - https://logs.px4.io/plot_app?log=d69f0576-9a6f-468f-820b-78af9c5b3264
Log of master - https://logs.px4.io/plot_app?log=1392a9b4-48d5-4f2b-be14-8f0720ce4d59

[image]

**Drone (please complete the following information):**
- VTOL design NXP & FliteTest.
- Photo of the IMU / autopilot setup if possible.
[image]

**Additional context**
Airspeed sensor is working on an old NXP_FMUK66_v2 board with FW of 2018.
Sensor was connected to 3 different NXP_FMUK66_v3 with different FW states (master, stable 1.9.0, stable 1.10.1) but none recognized the sensor.
Airspeed sensor was working on NXP_FMUK66_v3 20th of Nov 2019. Something might get broken afterwards.
I2C LED board is working on the I2C/NFC connector.

davids5, 2020-01-29:
@dk7xe the ms5611 is a baro. The Holybro sensor is an MS4525. Have you tried staring that driver `ms4525_airspeed start -a` ? The code is condition on fixed wing and vtol to start the driver [here](https://github.com/PX4/Firmware/blob/master/ROMFS/px4fmu_common/init.d/rc.sensors#L23-L45) 
all the changes in that code are years ago. If you send me the sensor I can help debug it.

davids5, 2020-01-29:
@dk7xe You can troubleshoot it with `i2cdetect -b <bus number>` It should show up when blogged in. at 0x28.

`#define I2C_ADDRESS_MS4525DO	0x28	/**< 7-bit address. Depends on the order code (this is for code "I") */`

dk7xe, 2020-01-29:
Hi David,

result of i2cdetect
[image]

and airspeed start
[image]

It looks like that the driver is not started automatically. And yes it's a VTOL airframe.

dagar, 2020-01-29:
The airspeed circuit breaker parameter  CBRK_AIRSPD_CHK` is set. Can you reset it (set to 0), reboot, and try again?

dk7xe, 2020-01-30:
Setting CBRK_AIRSPD_CHK back to 0 solved the problem.
Either this needs to be described in the user documentation (at least a warning) OR CBRK_AIRSSPD_CHK should not prevent having the sensor in use if present.

dk7xe, 2020-01-30:
But now the next problem arised.
After clicking on **Calibrate Airspeed [OK]** i get the following
[image]

dk7xe, 2020-01-30:
i have closed this now because the error is solved for me. But please update documentation accordingly.

hamishwillee, 2020-01-30:
Hi @dk7xe , @dagar 

Are we sure this is not a code bug?

The circuit breaker should remove the airspeed sensor prearm **checks**, but I would expect the sensor still to be started, and that if it can be started for it to be possible that it is calibrated. 

@dagar can you confirm what the circuit breaker should do?

If this is not a code bug we might update param docs as follows:

> Circuit breaker for airspeed sensor
>
> Comment: Setting this parameter to 162128 will disable the check for an airspeed sensor. The sensor driver will not be started and no calibration can therefore be done.
> WARNING: ENABLING THIS CIRCUIT BREAKER IS AT OWN RISK

dagar, 2020-01-31:
> Are we sure this is not a code bug?

Debatable perhaps, but currently a feature.

hamishwillee, 2020-01-31:
OK. I'll leave the code debate and any QGC update for you. Yes, perhaps we should have an airspeed sensor status in https://mavlink.io/en/messages/common.html#MAV_SYS_STATUS_SENSOR

 @dk7xe are you happy with proposed docs update?

> Circuit breaker for airspeed sensor
> 
> Comment: Setting this parameter to 162128 will disable the check for an airspeed sensor. The sensor driver will not be started and no calibration can therefore be done.
> WARNING: ENABLING THIS CIRCUIT BREAKER IS AT OWN RISK

hamishwillee, 2020-02-04:
OK. PR created for the parameter. I also updated vtol without airspeed sensor doc: https://github.com/PX4/px4_user_guide/commit/44b5efb428b6e0c820bf348a1d0ddf8c73edb1de

closed 2020-02-04 by LorenzMeier

dk7xe, 2020-02-04:
It's not a bug, it's a feature.. nice. *sarcasm off* ;)