PX4/PX4-Autopilot #14068: Huge amount of drifting on IRIS in SITL simulation

shrit, 2020-01-30:
Hello,

I am using IRIS for SITL simulation, I am using 4 quadrotors at the same time, all of them use the same PX4 version. Only one of them is facing a huge amount of drifting during the flight. 
Here is the last flight with this quadrotor
https://logs.px4.io/plot_app?log=460f767c-25fe-4702-b910-66f3645203af
You can notice a spiral landing which I have been observing only with this one. 

Any thought?
Best regards,

julianoes, 2020-02-12:
What could be different with number 4 than the others?

shrit, 2020-02-12:
@julianoes I have no idea, this is the reason I posted this issue. 
They all use the same version of PX4. The only difference is the model name and the port number in the SDF file.
Here is the log for another quadrotor that was behaving perfectly:
https://logs.px4.io/plot_app?log=00c4e2f7-a818-4b90-9a0f-9a934adf9625

Let me know what do you think

julianoes, 2020-02-14:
@Jaeyoung-Lim maybe you have an idea.

Jaeyoung-Lim, 2020-02-14:
@shrit How are you starting the multivehicle SITL simulation?

shrit, 2020-02-14:
@Jaeyoung-Lim I have extended the script in https://github.com/PX4/Firmware/blob/master/Tools/sitl_multiple_run.sh to start gazebo with the quadrotors.
Then I have written software to control multi-vehicles. Basically I take off the swarm at the same time, using several threads and I wait for those the processes to finish. After takeoff, switch everyone to offboard mode, I play around with the swarm. When the objective is reached, I land all of them and reset all of them using the reset plugin described in https://github.com/PX4/sitl_gazebo/pull/380
Note that, So far, I have noticed this problem in v1.9.2 of the firmware. I have switched recently to the last stable (1.10.1), everything seems to be fine (until now)

Jaeyoung-Lim, 2020-02-14:
@shrit Could you maybe give it a go with this script? https://github.com/PX4/Firmware/blob/master/Tools/gazebo_sitl_multiple_run.sh

The usage is documented here: http://dev.px4.io/master/en/simulation/multi-vehicle-simulation.html

I think we discussed this before, but resetting the model position while the EKF is running might make the EKF2 not recover from the sudden change of position

shrit, 2020-02-14:
@Jaeyoung-Lim Ok, I will give this script a try. I suppose it will generate automatically the necessary sdf file on any vehicle model.
Considering resetting the quadrotors I totally agree, but I have no other solution since it is not possible to reboot the firmware without stopping the entire simulation.
I found it strange that EFK2 does not behave similarly with all the quadrotors since there is only one quadrotor that is drifting.  
I have looked rapidly into documentation, the video for IRIS quadrotors seems to suffer from high real-time factor decreasing, Did you initially changed the real-time factor? or it has decreased during the simulation? since I run the simulation for days I am afraid I might be getting the even worse result.
In fact, I am facing this decreased real-time factor, but it takes several hours to decrease since I am using only 4 quadrotors.

julianoes, 2020-05-15:
@shrit is this still an issue?

shrit, 2020-05-15:
@julianoes I have only noticed this issue in px4 v1.9.2.
Never occurred (So far) in 1.10.1.
Maybe we can close it for now

closed 2020-05-15 by Jaeyoung-Lim