PX4/PX4-Autopilot #14069: HIL: disable lockdown not working

gian1312, 2020-01-30:
**Describe the bug**
Same behavior as in [#5680](https://github.com/PX4/Firmware/issues/5680).
When disabling the lockdown with the `px4io lockdown disable` command, there is no output to the real actuators in HIL-mode.

**To Reproduce**
Steps to reproduce the behavior:
1. Power Servo Rail and make sure that all CS and props can move freely (and the plane is secured) 
2. Start in Hil-Mode
3. Arm Plane
4. Enter said command
5. Move your TX (Idealy something would happen)

**Expected behavior**
I want the actuators to move in Gazebo and on the real plane simultaneously.
At the moment, they work either in HIL in Gazebo or in reality with HILL disabled.

**Log Files and Screenshots**
There are no log files. If there is anything, I could provide, please let me know.

**Drone (please complete the following information):**
Fixed Wing (MPX Easy Glider)

gian1312, 2020-02-20:
Are there any inputs/ideas how I could fix this?

julianoes, 2020-02-20:
I think you would have to debug (using a console connected and printf) your way through all the lockdown flags, or if it's correct on the FMU side, inspect the pxi4ofirmware code:
https://github.com/PX4/Firmware/tree/master/src/modules/px4iofirmware

gian1312, 2020-05-20:
Has it been addressed yet?

Mjohn1108, 2020-06-18:
yeah, same problem here i want to do HIL simulation with the servos and motors  actually running. With piccolo servos and motors are working perfectly in HIL. We want to see the actual behavior of the plane.

HashimAifi, 2020-11-05:
I got it working. This is what I did:
1) I used the latest px4 firmware (v1.11.1 - also works on current master branch)
2) I only changed one file (state_machine_helper.cpp): Commented the line `armed->lockdown = true;` inside `if (hil_enabled) {` block
3) I compiled, built and then uploaded this firmware in QGC v3.5.6
4) In QGC I **disabled** HITL then rebooted vehicle and then selected Standard-Plane as the airframe
5) I pressed the safety switch to check if my servos are working as per the radio control.
6) Then, I **enabled** HITL, again rebooted the vehicle and selected 'HIL-Star' as my airframe.

Note: At this point, the servos should still be active and controlled by radio.

7) I launched XPlane-11 and started a flight
8) In QGC, I opened the HIL widget to connect with X-Plane 11
9) Then, I armed the vehicle

After arming; the real servos and the X-Plane aircraft wings should both be working and you should also be able to fly the plane using the radio.

I hope this helps someone.

MuneebaAsif1, 2024-02-29:
Honestly, I keep seeing this one comment regarding v 1.11.1 everywhere, but it does NOT work. Does anyone have a working solution?

julianoes, 2024-02-29:
v1.11.1 is ancient. I doubt anyone is going to help with that.

MuneebaAsif1, 2024-02-29:
Yes, I am using 1.13.3 with QGC v4.3.0. How do I enable motor actuation with HITL that I can observe on both the hardware and jMAVsim?

NanyongGit, 2024-03-14:
Looking for a solution as well to get the motors running... Anyone got anything?