PX4/PX4-Autopilot #14070: Reboot is killing the entire SITL simulation

shrit, 2020-01-30:
Hello,

Starting simulation using make `px4_sitl gazebo_iris `I am trying to reboot by using `reboot` command in `px4>` shell

However, the entire simulation is terminating as if I had asked for shutdown instead of reboot 
Is this normal?

shrit, 2020-01-31:
I suppose that this shell does not work in SITL simulation as it requires Nuttx to be used instead of Linux

julianoes, 2020-02-04:
This is "normal" but I agree that it would be nice if this was not implemented that way. As far as I remember, currently we are not able to do a controlled clean shutdown, so all we can do is `exit`.

I think the first step will be to implement a clean shutdown which then would allow reboots.

See #11654.

shrit, 2020-02-04:
Ok, I see.
I understand that this is normal, I had a glance on the code, I do not have any experience in operating system programming so I do not know how can I help. 
I do not know when this feature might be ready, so at least we have it know at the list of requested features.

shrit, 2020-05-09:
Keep open

shrit, 2020-08-08:
Keep open