PX4/PX4-Autopilot #14071: mc_pos_control: thrust limits

dusan19, 2020-01-30:
In 1.10, mc_pos_controller does not always limit the thrust setpoint which is passed to the attitude controller.

In situations:
- No flight task active
- Idle flight task running
- ground contact or maybe landed
The thrust is not the minimal thrust set with the parameter (MPC_THR_MIN), but is 0

Is this behavior intentional?
When I arm the drone and set it to hold mode (before I run the mission) the props should spin with minimal specified thrust not with 0, same holds when ground contact is detected, the thrust shouldn't drop all the way to 0 in my opinion.

bresch, 2020-02-01:
The minimum thrust defined by MPC_THR_MIN is intended to be the minimum thrust during flight to avoid loosing attitude authority when "airmode" is disabled. On the ground, we just want  to start the motors, no attitude control is required and this is why the "idle" value (0) is used. 

Could you describe what is the you are trying to fix? Is it just a documentation issue?

dusan19, 2020-02-01:
Thank you for the answer. It was more of a FMI question, probably the issue was a wrong place to ask.
The reason I asked is that I noticed lately that while landing the drone cuts off the thrust significantly more when detecting land. It turned out to be for a different reason.

closed 2020-02-01 by dusan19