PX4/PX4-Autopilot #14075: PX4 10.0.1 Stable, Hard Fault in file:armv7-m/up_hardfault.c

Chris-Bee, 2020-01-31:
Hello,
I encountered an issue with PX4 1.10.1 Stable where the copter seems to make a 180° flip upside down. I am using a MoCap pose input over MavRos, mapped to the vision pose topic as described in the [tutorial](https://dev.px4.io/v1.10/en/ros/external_position_estimation.html ). The logs indicate that this seems to be a software crash, thus the movement might be a random result due to the underlying error. The Pixhawk 4 gives multiple beeper signals after engaging the killswitch.
Both logfiles below show 'Type:Hard Fault in file:armv7-m/up_hardfault.c at line: 148 running task: init'

**To Reproduce**
Steps to reproduce the behavior:
1. Map the MoCap pose to the vision pose topic
2. Enable the vision pose estimate
3. Start the copter in altitude mode and take off
4. Switch to position mode
5. The copter holds its position stable for ~5 seconds until the described issue occurs 

**Expected behavior**
The copter should hold its position.

**Log Files**
MoCap at 120Hz, the log indicates I2C3 low on the stack:
https://logs.px4.io/plot_app?log=92b60ea2-4b1e-4fac-9402-e1c797494866
MoCap at 30Hz, the log does not point out a clear error but the same issue occurred:
https://logs.px4.io/plot_app?log=afa24f21-0f51-42a4-bd2e-9b90ef29b14c

**Hardware**
Pixhawk 4 Holybro

dagar, 2020-01-31:
Do you have any "good" logs from this setup? Were you using an older version of PX4 successfully?

dagar, 2020-01-31:
By any chance are you using the console when this happens?

Chris-Bee, 2020-01-31:
Hello @dagar,
thank you for the fast reply.

> Do you have any "good" logs from this setup? Were you using an older version of PX4 successfully?

Unfortunately, I do not have logs with a working setup prior to 1.10.1 Stable, but I can test 1.10.0 on Monday and upload the logs

> By any chance are you using the console when this happens?

The consol was not connected, but this can also be done on Monday. Should I observe something specific with the console?

dagar, 2020-01-31:
> Unfortunately, I do not have logs with a working setup prior to 1.10.1 Stable, but I can test 1.10.0 on Monday and upload the logs

I don't think that's necessary, I was mainly wondering if this had ever worked entirely at some point and we could review the log.

> The consol was not connected, but this can also be done on Monday. Should I observe something specific with the console?

I saw mentions of the shell (nsh) in the hardfault, so i was wondering if this might have corresponded with any particular input (this includes the mavlink shell).

From your perspective did the hardfaults occur at the same phase of flight?

Chris-Bee, 2020-01-31:
> I saw mentions of the shell (nsh) in the hardfault, so i was wondering if this might have corresponded with any particular input (this includes the mavlink shell).

If this includes mavlink then yes, the console was connected. The MoCap pose is communicated via MavROS to px4.

> From your perspective did the hardfaults occur at the same phase of flight?

Yes, the scenario was comparable. The vehicle was in altitude mode in the beginning, and after switching to position control, it took about 5 seconds till the error occurred both times. The first flight should have inputs from the RC during position mode and for the second flight, the vehicle did just hover without any further RC input in position mode.

Chris-Bee, 2020-03-09:
Hello @dagar,
are there any updates from your side regarding this issue? We are planning to downgrade to 1.10.0 stable next week and test if the error occurs for this version as well.

dagar, 2020-03-09:
> Hello @dagar,
> are there any updates from your side regarding this issue? We are planning to downgrade to 1.10.0 stable next week and test if the error occurs for this version as well.

Would you be able to quickly try master (what's going to become v1.11.0 soon)? If you're able to reproduce the failure there it might help if we can setup a call and debug together.

koenlucassen, 2020-05-20:
I have actually come across the same error. A link to the log can be found below:

koenlucassen, 2020-05-20:
Hello everyone,
I seem to have the same problem, the software crashes shorty after take-off with a fixed-wing airplane using a black HEX cube. The log file can be found below:
https://logs.px4.io/plot_app?log=8572edab-7b45-4eb3-8ae8-d6a7036de909

closed 2020-12-29 by mhkabir