PX4/PX4-Autopilot #14077: Can't build with python3

fnoop, 2020-01-31:
PX4 fails to build with a custom version of python3.

```
-- Detecting C compile features - done
-- Found PythonInterp: /usr/bin/python (found version "2.7.17")
CMake Error at /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  couldn't find python module jinja2:

  for debian systems try:

  	sudo apt-get install python-jinja2

  or for all other OSs/debian:

  	sudo -H pip install jinja2

   (missing: PY_JINJA2)
Call Stack (most recent call first):
  /usr/share/cmake-3.10/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  cmake/px4_find_python_module.cmake:61 (find_package_handle_standard_args)
  CMakeLists.txt:276 (px4_find_python_module)

-- Configuring incomplete, errors occurred!
See also "/srv/maverick/code/px4/build/px4_sitl_default/CMakeFiles/CMakeOutput.log".
Error: /srv/maverick/code/px4/build/px4_sitl_default is not a directory
Makefile:193: recipe for target 'px4_sitl_default' failed
make: *** [px4_sitl_default] Error 1
```

This has detected system python2 which is not in use - instead I have a custom version of python3 installed that is used for all python on this system (in /srv/maverick/software/python/bin).  System path contains python3 before python.  In addition, PYTHON_EXECUTABLE is also set:
`PYTHON_EXECUTABLE=/srv/maverick/software/python/bin/python3`

How to specify the python interpreter?

fnoop, 2020-01-31:
Note if I change CMakeLists.txt from:
`find_package(PythonInterp REQUIRED)`
 to
`find_package(PythonInterp 3.5 REQUIRED)`

Then it finds my python3 interpreter and works.

Also, if I change Makefile to add:
`CMAKE_ARGS += -DPYTHON_EXECUTABLE=/srv/maverick/software/python/bin/python3`
Then it also works

fnoop, 2020-01-31:
If this is added to Makefile, then at least we can set PYTHON_EXECUTABLE environment variable to hint the compile:
```
ifdef PYTHON_EXECUTABLE
	CMAKE_ARGS += -DPYTHON_EXECUTABLE=${PYTHON_EXECUTABLE}
endif
```

FlavioTonelli, 2020-01-31:
This shall be already fixed in master. Give it a try

fnoop, 2020-01-31:
Hi @FlavioTonelli - Ah yes I can see that CMakeLists.txt now requires python3 as minimum requirement for non-catkin build.  However you still can't specify/override the python interpreter, it has to be the priority in the system path.
OK if I make a PR for the above fix?

FlavioTonelli, 2020-01-31:
I'm not the maintainer of the code so I can't provide you a detailed feedback on this.
On my point of you the improvement make sense. Feel free to contribute :)

fnoop, 2020-01-31:
PR https://github.com/PX4/Firmware/pull/14080 raised

julianoes, 2020-02-03:
@fnoop I'm confused. Why isn't this used:

https://github.com/PX4/Firmware/blob/86eff44a5efa01273788bc46e0001c4102cafff2/CMakeLists.txt#L270

fnoop, 2020-02-03:
@julianoes `Python Interp 3` was only added after the latest release - I was using 1.10.  #14080 allows us to specify whichever interpreter we want so improves this, thanks for approving :)

closed 2020-02-03 by fnoop