PX4/PX4-Autopilot #14081: gyro 1 (ID 682249) no matching uORB devid

bilginmu, 2020-01-31:
I am using Pixhawk Cube. During the gyroscope calibration, I get an error like below and calibration does not end.

gyro 1 (ID 682249) no matching uORB devid

dagar, 2020-01-31:
Which version of PX4? When did you last calibrate?

bilginmu, 2020-01-31:
I think it is last version of PX4. Today, I installed PX4 firmware with QGC and calibrated it. Additionally, I am using external GPS (GPS Here). I can calibrate well without GPS but cannot calibrate with external GPS.

bilginmu, 2020-02-01:
I recognize another error today.

- Calibrate compass
- "Mag 2 Fail: STALE!" error comes up during compass calibration
- Calibrate gyroscope 
-  "gyro 1 (ID 682249) no matching uORB devid" and "Mag 2 Fail: STALE!" errors comes up during gyroscope calibration. 

How can fix it ? 

Thanks.

bilginmu, 2020-02-04:
When I changed my firmware(stable v1.10.1) to developer version, problem is fixed.

closed 2020-02-04 by bilginmu

bigbellmercy, 2020-05-06:
The same error happened with Pixhawk2.1 + Here GPS. From PX4 v1.11.0-beta1, the error is disappeared but it happens with PX4 v1.10.2 and v1.10.1.

koenlucassen, 2020-05-06:
Yeah same here with the Hex Cube, when the GPS (external with compass) is connected. I have exactly the same as posted here: https://discuss.px4.io/t/gyro-1-no-matching-uorb-devid/14952.

First you get the MAG Stale error. Magnetometer calibration still works, but gyroscope and accelerometer calibration fail. Is this error caused by the introduction of the a new external compass/magnetometer?

Would appreciate a solution to this problem.

koenlucassen, 2020-05-06:
I have done a bit of digging and I think this is what goes wrong. The external GPS also has a 9DOF IMU, for me that is the icm20948. This sensor is started before the other imu board sensors. This means that the external IMU is the zero instance.

Now during the calibration the max gyro and accel count is 3, while if you are using an external GPS/compass this should be 4. The check in the calibration checks for the driver_id and the driver_id given in the report (uorb message). Having 4 active gyro/accels and only 3 that are checked gives for one device an error.

What for me currently works is commenting out the initialization of the icm20948 sensor, as I do not want to use these external sensors. Probably a better solution is to increase the count of accel/gyros or start the icm20948 after the other imu board sensors.

dagar, 2020-05-06:
Thanks for sharing the details @koenlucassen. We could fairly easily extend the system to support 4 accels/gyros, but I actually don't think we even want to use the external IMU on the icm20948. It's attached on i2c and not capable of the high rate sampling we currently do by default with Invensense IMUs.

Have you tried running the ak09916 (the mag within the icm20948) driver directly? If that doesn't work I'll get the hardware and take a closer look.

koenlucassen, 2020-05-06:
Yes that makes sense! Yeah so for me it works to only run the ak09916. It also works when you initialize the icm20948 after all the other imu's on the board. I think you achieve the same thing here, as the accelerometer and gyroscope still do not show up in uorb messages of the gyro or accel.