PX4/PX4-Autopilot #14088: Fixed wing position control code documentation

prabhjeetsingh88, 2020-02-03:
Hi, I am a newbie here. I am trying to understand fixed wing position control code for PX4 firmware.
But I am unable to find the documentation which explains this and other code in detail. Could you please send me the URL for the same.

Any help in this regard will be highly appreciated.

Thanks

julianoes, 2020-02-04:
I think all there is, is the actual code:

https://github.com/PX4/Firmware/tree/master/src/modules/fw_pos_control_l1

https://github.com/PX4/Firmware/blob/ca96b6b0eae2d246045e2e85f52414dd1b14dd0c/src/modules/fw_pos_control_l1/FixedwingPositionControl.cpp#L539-L541

For altitude/speed/pitch control there is TECS:
https://github.com/PX4/ecl/tree/master/tecs

For more general documentation, I'm sure you have seen the devguide:
https://dev.px4.io/master/en/concept/architecture.html

hamishwillee, 2020-05-06:
Is this useful? http://dev.px4.io/master/en/flight_stack/controller_diagrams.html#fixed-wing-position-controller

bresch, 2020-08-10:
I think the fixedwing documentation is quite good now.

closed 2020-08-10 by bresch