PX4/PX4-Autopilot #14094: feature request : Hybrid LAND mode for VTOL UAVs

moreba1, 2020-02-03:
Hi

When the UAV is flying at a high altitude (400 m) and you want to land the UAV there . power consumption is very hight  and the battery may not be sufficient. In my opinion, it is better to add a hybrid flight mode in code to lower the height in circular loiter to 40 m first, then transtiotion to multi-rotor mode and descend.

julianoes, 2020-02-04:
I agree. However, I would argue that this could generally be taken care of by a mission with planned landing sequence.

@sfuhrer might be able to better comment.

moreba1, 2020-02-04:
it can not done by a planned mission because we will land emergency in that position .

sfuhrer, 2020-02-04:
Yes @julianoes is right, this problem can be solved by uploading a mission with a land_start marker, then a loiter to altitude followed by the land wp. When you then trigger a RTL (or it get's triggered by safety mechanisms) it will fly the mission from the land_start marker, and transitions at a low altitude (the one that's set in the loiter to alt wp). 
RTL_TYPE = 1 or 3 required.

This way works though only for one landing spot, eg not for rally points. Happy for contributions to improve that!

moreba1, 2020-02-04:
@sfuhrer  RTL application is different with land. when battery is too low for RTL or in GPS loss , RTL by mission is not a good choice.

hamishwillee, 2020-02-04:
Just to clarify current behaviour, land mode is documented [here](https://docs.px4.io/master/en/flight_modes/land.html#vtol). According to this the vehicle will start landing according to current mode. If that mode is [Fixed Wing mode](https://docs.px4.io/master/en/flying/fixed_wing_landing.html) then there is a setting that causes it to switch to MC mode just before landing [NAV_FORCE_VT](https://docs.px4.io/master/en/advanced_config/parameter_reference.html#NAV_FORCE_VT) that is default on. In RTL this is not configurable - the landing always switches to MC.

@julianoes - What does "just before landing" actually mean? Ie at what height does it switch.

So the request is that you want a land mode for VTOL where:
- If in FW mode you circle down to a specified height (40m), transition to MC and then land
- In MC mode you want to transition to FW mode, go down to specified height, then transition back and land?
Is that correct?

The FW mode would seem to be much the same as what we have, if the transition height is configurable. Though it is an FW landing rather than circular loiter - what is the benefit of loiter here? 
The MC mode a bit different, but you're unlikely to be flying MC at that height anyway right?