PX4/PX4-Autopilot #14101: Cannot start EKF2 on External Vision without a Barometer

marcelino-pensa, 2020-02-04:
**Describe the bug**
EKF2 does not start if there is no incoming barometer data, even with `SYS_HAS_BARO = 0`. This seems like a bug to me unless this is expected behavior for some reason.

**To Reproduce**
I am running on a snapdragon flight using Px4 1.8, but it seems like this is a bug on master too (at least it seems so by looking at the code).
- Do not start the barometer when starting px4
- Use EKF2 enabling External Vision and disabling the barometer:
```
param set EKF2_AID_MASK 24
param set EKF2_HGT_MODE 3
param set SYS_MC_EST_GROUP 2
param set SYS_HAS_BARO 0
```
- Launch px4, EKF2 does not start

**Expected behavior**
I would expect EKF2 to start if SYS_HAS_BARO is set to 0. I know that EKF2 does not start because when I enable the barometer back again, the following log shows up:
```
pxh> INFO  [ecl/EKF] EKF aligned, (EV height, IMU buf: 26, OBS buf: 16)
INFO  [ecl/EKF] EKF commencing external vision position fusion
INFO  [ecl/EKF] EKF commencing external vision yaw fusion
```

**Drone (please complete the following information):**
- Snapdragon flight
- Photo of the IMU / autopilot setup if possible.

**Additional context**
It seems like the problem is in line 177 of [ecl/ekf.cpp](https://github.com/PX4/ecl/blob/a3e1eb9d509d55c2c476ec093cf24d84da9798e5/EKF/ekf.cpp). If there are not enough initial barometer measurements, the function `initialiseFilter()` returns false, which in turn does not allow EKF2 to execute.

LorenzMeier, 2020-02-04:
Many thanks for the detailed report. However, this is really a feature request for a very specific use case. Could you please send a pull request with the required changes instead? Open source projects are always open to accept new features that match coding standards and quality, but the way this works is that we have to rely on the people who need those to implement and test them.

https://help.github.com/en/github/collaborating-with-issues-and-pull-requests/about-pull-requests

marcelino-pensa, 2020-02-06:
Thanks for the prompt response, @LorenzMeier! I am willing to make the changes and test, but I am planning to do so on v1.8.0, since this is the one I am currently using in our project. I'm not sure if that helps at all, since this would require tests on master too.
I guess that the sole existence of this issue (even if we close it) can provide some guidance to other people who might encounter the same problem in the future (with recommendations on how to fix it too).
BTW, the correction needs to be done on [ECL](https://github.com/PX4/ecl/tree/master/EKF) repository, not on Firmware. Would it make sense to movie this issue there?

bresch, 2020-05-13:
Same as https://github.com/PX4/ecl/issues/800. EKF2 currently needs baro and mag data to initialize even if it doesn't use those.

closed 2020-05-13 by bresch