PX4/PX4-Autopilot #14103: Error while executing MAKE PX4_SITL GAZEBO (error: ‘TRUE’ was not declared in this scope.)

abdul-baais, 2020-02-05:
While i am trying to execute the MAKE PX4_SITL GAZEBO , it throws the following error.
Firmware/Tools/sitl_gazebo/include/gazebo_opticalflow_plugin.h:43:18: error: ‘TRUE’ was not declared in this scope.
and i am clueless, thanks in advance

TSC21, 2020-02-05:
Fix was brought here: https://github.com/PX4/sitl_gazebo/pull/408. sitl_gazebo submodule just needs to be updated in PX4 Firmware upstream.

closed 2020-02-05 by abdul-baais

mzahana, 2020-03-07:
I am still getting the same error.
* Using v1.10.1
* Did `git submodule init && git submodule update --recursive`
* `make px4_sitl gazebo`

Still getting the same error as of the time of this reply.

@TSC21 any hints?
Thanks.

marcelino-pensa, 2020-04-22:
You can fix it manually by:
Open `~/src/Firmware/Tools/sitl_gazebo/include/gazebo_opticalflow_plugin.h` and replace `#define HAS_GYRO TRUE` with `#define HAS_GYRO true` (make TRUE) lowercase.

It seems like the submodule related to the tag `v1.10.1` does not have the update that fixes this.