PX4/PX4-Autopilot #14105: PWM_MAX setting to 2200us only outputs 2150us

ataits, 2020-02-05:
**Describe the bug**
When I set the parameter PWM_MAX to 2200us, I expect that the maximum output should be 2200us. However, when I check the port0 outputs via QGC analyzer, I see only 2150us at maximum throttle (in acro mode).

**To Reproduce**
Steps to reproduce the behavior:
1. Drone attached to computer (QGC) via USB
2. Enable Circuit Breaker for Drone to Arm with USB
3. Set the PWM MAIN MAX parameter to 2200us in QGC
4. Arm the Drone in Acro Mode
5. Turn on QGC Analyzer showing port0 PWM outputs
6. Set Throttle stick to maximum
7. Observe PWM output only goes up to 2150us

**Expected behavior**
PWM output should go up to 2200us at maximum throttle

**Drone**
- QAV250 with Pixhawk 4

**Additional context**
The idea to enable full range output of 800us to 2200us. This is very important for ESC and on other platforms especially servos that can benefit from full amplitude of motion.

dagar, 2020-02-05:
Hi @ataits 

The limit is hard coded here. https://github.com/PX4/Firmware/blob/37db7d3bba965c379ac5e997661fa4930783daf3/src/drivers/drv_pwm_output.h#L101

Is this something we need to change? At the moment I'm not aware of a reason it should be 2150 us vs 2200 us.

coderkalyan, 2020-02-07:
Correct me if I'm wrong; Not trying to discourage you from picking what value you want as you probably have good reason for doing it, but I believe the standard for drone-related PWM control is a range from 1000-2000 us... Is there a particular effect you are trying to achieve by setting it to 2200 us?

feemi, 2020-02-07:
The problem is parameter reference is not consistent with this hard coded value. It says maximum value  can be set to 2200. In most cases. 1000 2000 is fine for standard escs. But wider ranges can work for servo travels.  Any way it doesnt make sense when hard limit is less than parameter max value.

julianoes, 2020-02-12:
> Any way it doesnt make sense when hard limit is less than parameter max value.

Agreed. Could you make a PR to fix it? Thx.

ataits, 2020-02-14:
Should I make a PR to fix this? Can we include this fix in the upcoming 1.11.0 release?

julianoes, 2020-02-20:
Yes please, and I'd think that should make it in.

taileron, 2021-04-17:
There are always the correct default limits in the corresponding settings.
Then a default limit is also entered in the driver.
Why is highest max limited in the driver at all?
That is the main reason that I cannot use pre-compiled binaries even for testing.

In the PWM output driver, the highest degree of freedom should be set to highest max = 2500uS.
Then it will be reduced later by the default or you can enter something suitable for the individual project.