PX4/PX4-Autopilot #14116: PX4IO not enough RAM left for VTOL models like convergence

julianoes, 2020-02-06:
@sfuhrer found that sometimes RC did not work on the convergence and that the ACT LED on the Pixhawk 4 IO side would be always on instead of blinking.

We could also see a segfault using GDB via Dronecode Probe:

TODO: @sfuhrer fill in backtrace

According to `git bisect` the bytes free go from 568 to 88 (sometimes 24) anywhere here, presumably due to the NuttX upgrade:

The first bad commit could be any of:
58799dc7d17b01e4e586463b23cb83692b678b77
694ae56afea4fdcd7d8ef665ab7a516e758dd7df
e847698c9f0ec950310eea30313340bf98b37df0
36a75b79a418129913936bd987717afa8914a911
645b6c3fbcdeb2b5f9d3e56807fbaa1fa74811ec
0f2a7ad2b7d2acea354f1b91771512f2a1f76eee
522ed2d99ba4b0ed6f749a6c2218a6d4ec0c9bef
78944a70c345d1da51e906e6225352607675ac21
d965b928f8ef4985c0799e8ce64df4d9c4912a2b
69694ab000f923106931612fa34f8d689b1732fc

*Steps to reproduce:*

Set `SYS_AUTOSTART` to 13012 for convergence airframe.
Boot and do `px4io status`.
You should see:
```
88 bytes free
```
Or sometimes garbage values when it crashed.

FYI @bkueng @davids5.

dagar, 2020-02-06:
For comparison.

``` Console
PX4IO: starting
MEM: fre` 1160, Hargest 1160  <-- stable v1.10
MEM: free 622, largest 632    <-- master
```

julianoes, 2020-02-06:
@dagar What is that comparing? And what airframe?

dagar, 2020-02-06:
It's the SYSLOG output in px4io.c before the mixer is even loaded. 

https://github.com/PX4/Firmware/blob/b399698306e5afa068903adda094adedf0423691/src/modules/px4iofirmware/px4io.c#L359

davids5, 2020-02-06:
@julianoes - it was the last upgrade of NuttX, when the H7 came in. FD/streams became required.  We could set it to 0 in the past and that saved memory.  I adjusted the stack and free mem to boot. with out police lights, but Not all the mixers were tested. :(

sfuhrer, 2020-02-07:
Hi guys, do you see an easy way to unblock people that are using this specific airframe?

julianoes, 2020-02-10:
@sfuhrer #14128 might be just enough for now.

sfuhrer, 2020-02-10:
> @sfuhrer #14128 might be just enough for now.

True, the IO with the Convergence works again with this PR (with 40 bytes free).

dagar, 2020-05-06:
This was fixed, although we're still slightly short for the most extreme case (dodecarotor coax).

closed 2020-05-06 by dagar