PX4/PX4-Autopilot #14127: attitude estimator not running - check system boo and Pref-fight checks failed Baro Sensors 0 Missing

majid55-pixmini, 2020-02-09:
i have pixhwak 4 mini fc and recently it is giving me these errors.
1-attitude estimator not running - check system boo
2-Pref-fight checks failed Baro Sensors 0 Missing
board have latest px4_fmu-v5 V1.10.1 stable version. updated through QGC
and then i tried to reinstall previous versions too but error doesn't go away.
i did not crash my drone and it was working good before installing latest version.
if i disable ekf2 in perimeters.
and press the safty switch on GPS i am able to arm and fly the drone but it is not holding altitude anymore and if i flip the switch for alt hold or GPS hold it say "Reject altitude Hold " Reject GPS hold "
Alt hold and GPS lock was working before only new hardware i added is telemetry device and updated to latest V1.10.1.
so what could be the reason i am sure it is software reason because when i google this error lot's of people have this issue after updating the software so is there any fix for this issue.
Thanks i advance

julianoes, 2020-02-12:
Could you please upload a log to https://logs.px4.io/ ?

sahaaswina, 2020-03-05:
I have exactly same problem, never crash, no new hardware, just update firmware and get same message, I have try re-flash and reset and still not working. What log do you need?

SalimTerryLi, 2020-03-05:
> I have exactly same problem, never crash, no new hardware, just update firmware and get same message, I have try re-flash and reset and still not working. What log do you need?

Can you test over nsh to figure out whether ms5611 driver is working or not? Please try to execute 
`ms5611 status` and `ms5611 -s start`

sahaaswina, 2020-03-05:
`nsh> ms5611 status`
`WARN [ms5611] driver not running`
`nsh>`
`ms5611 -s start`
`MS5611_SPI on SPI bus 4 at 0 (20000 KHz)`
`WARN [ms5611] no device on bus 3`
`nsh>`
`ms5611 status`
`WARN [ms5611] driver not running`
`nsh>`

hope this is what you expected

SalimTerryLi, 2020-03-06:
It seems at least one of these two calls failed.
https://github.com/PX4/Firmware/blob/4698a09b987428b5d26928935e224a47b827e77e/src/drivers/barometer/ms5611/ms5611_spi.cpp#L134-L147
@davids5 Can you please explain how to get the output of `PX4_DEBUG` here? We need those message to locate the error.

davids5, 2020-03-06:
@SalimTerryLi  in the driver constructor add `_debug_enabled= true;`

davids5, 2020-03-06:
If that does not work, then check the macros in log.h

`CXXFLAGS=-DDEBUG_BUILD make <board>`

Options are DEBUG_BUILD or TRACE_BUILD

@dagar , @hamishwillee is this documented?

hamishwillee, 2020-03-10:
@davids5 No, news to me.

I can see in https://github.com/PX4/Firmware/blob/master/platforms/common/include/px4_platform_common/log.h that different macros are compiled in for TRACE_BUILD or DEBUG_BUILD.

I think you are saying that *somewhere* (where?) you can put a line like `CXXFLAGS=-DDEBUG_BUILD make <board>` or  `CXXFLAGS=-DTRACE_BUILD make <board>` to enable those. 
- It isn't clear why the debug option wouldn't be on for a debug build by "default" 
- or when you would enable trace.

If you guys can propose some text that answers those questions then I can put it in.

dagar, 2020-03-10:
We typically don't do overall debug builds because there's just too much output. When needed I usually set `-DDEBUG_BUILD` per module/driver in px4_add_module COMPILE_FLAGS (CMakeLists.txt).

davids5, 2020-03-10:
@dagar can you post an example changed file here?

davids5, 2020-03-14:
@hamishwillee 

Here is an example provided by @dagar Please add an explanation it to docs.

[image]

hamishwillee, 2020-03-16:
@davids5 Since I know very little about this topic I can't help feeling you'd do a better job of this.

But here goes. Additional questions:
1. So what is the default kind of build for modules? RELEASE_BUILD?
2. Proposal is to add a heading in [logging](http://dev.px4.io/master/en/log/logging.html) topic called "Verbose Logging"? Something like the below OK?

---

## Verbose Logging

Modules output minimally verbose logs strings by default (e.g. for `PX4_DEBUG`, `PX4_WARN`, `PX4_ERR`, etc.).

Log verbosity is defined at build time using the `RELEASE_BUILD` (default), `DEBUG_BUILD` (verbose) or `TRACE_BUILD` (extremely verbose) macros.

Change the logging level using `COMPILE_FLAGS` in the module/driver `px4_add_module` function (**CMakeLists.txt**).
The code fragment below show the required change for the module template example ([src/templates/module/CMakeLists.txt](https://github.com/PX4/Firmware/blob/master/src/templates/module/CMakeLists.txt)) 

```
px4_add_module(
	MODULE templates__module
	MAIN module
```
```
	COMPILE_FLAGS
		-DDEBUG_BUILD
```
```
	SRCS
		module.cpp
	DEPENDS
		modules__uORB
	)
```

> **Caution** You can also change the log verbosity at board level, but this is not recommended as there is too much output: 
> ```
>  CXXFLAGS=-DDEBUG_BUILD make <board>
>  ```

davids5, 2020-03-16:
> So what is the default kind of build for modules? RELEASE_BUILD?

For a definitive answer you have to check the log on the build server and on a dev box and see what Cmake reports.

@dagar would you fill in the blanks from the Cmake type to what flag xxxx_BUILD get added to the 
build.

@hamishwillee - I think we need to share what works and will work for a developer. So unless you test what you are proposing, I would only share the info in a "driver debugging" section and leave it at that.

hamishwillee, 2020-03-17:
Thanks @davids5 . Fair enough, driver stuff in https://github.com/PX4/Devguide/pull/976 - please sanity check.

If you want the broader information please add it to the docs yourself and I can subedit.

cnpcshangbo, 2021-10-12:
I figured out how to solve this issue. Setting all the parameters to defaults would solve this issue. Here's a video for that:
https://umsystem.hosted.panopto.com/Panopto/Pages/Viewer.aspx?id=6325b151-50d9-464f-a1c3-adbf01415398

m4v3r, 2021-11-10:
Same issue on two seperate boards. Been switching around firmware versions (mostly 1.12.3 and 1.11.3) and sensors.

dagar, 2021-11-10:
> Same issue on two seperate boards. Been switching around firmware versions (mostly 1.12.3 and 1.11.3) and external sensors.

@m4v3r if you can share a log from boot with PX4 v1.12.3 or current master I'll take a look. https://logs.px4.io

m4v3r, 2021-11-10:
> > Same issue on two seperate boards. Been switching around firmware versions (mostly 1.12.3 and 1.11.3) and external sensors.
> 
> @m4v3r if you can share a log from boot with PX4 v1.12.3 or current master I'll take a look. https://logs.px4.io

Here are three different logs from one board, powered directly from a PSU, nothing attached except USB.

1: Gyro & Accel finish calibrating
[https://logs.px4.io/plot_app?log=2a524200-1e39-4058-9bdb-6f84833647c8](url)

2: Level-Horizon calibration fails with error "Attitude estimator nut running. check system boot..."
[https://logs.px4.io/plot_app?log=a0c5be56-8be5-4e49-bcb0-bcce90806479](url)

3: Gyro & Level-Horizon finish under Attitude Estimator Q
[https://logs.px4.io/plot_app?log=bd690831-5b0b-425e-a649-af81e9d7a487](url)

I noticed the altitude isn't showing up in QGC either but is publishing something under the altitude_monotonic topic but this might be a seperate issue even.
Thanks a lot for your time. ;)

copterspace, 2022-04-26:
> 2: Level-Horizon calibration fails with error "Attitude estimator nut running. check system boot..."

I encountered such a problem when I uploaded LPE firmware to pixhawk.
Before calibration need to set SYS_MC_EST_GROUP = LPE instead of default EKF2.
After restart attitude estimator starts working , and all level calibration proceeds OK.
[image]

m4v3r, 2023-03-11:
> I encountered such a problem when I uploaded LPE firmware to pixhawk. Before calibration need to set SYS_MC_EST_GROUP = LPE instead of default EKF2. After restart attitude estimator starts working , [...]

This didn't work on the board I'm working on (issue persisted with all possible estimator type settings, no level calibration possible and error still popping up), however, I managed to get it back running normally by flashing several times and in several ways with current master firmware.