PX4/PX4-Autopilot #14137: Periodic cutoff of local position from EKF2

ashwinsushil, 2020-02-10:
**Describe the bug**
I am observing periodic cutoff of local position data from the EKF2 with the setup as mentioned below. 

At the cut-offs, the optical flow is giving valid data.

**Setup**
1. Intel aero drone
2. PX4 v1.9.2 with PX4Flow driver enabled
3. Px4Flow
4. Parameters:
  * EKF2_AID_MASK = 2 (Optical flow)
  * EKF2_HGT_MODE = Barometer
 5. Teraranger One data sent from a computer companion to FCU through mavros.

**To Reproduce**
Steps to reproduce the behaviour:
1. Flash an intel aero drone with v1.9.2 PX4 firmware with PX4 driver enabled.
2. Connect PX4Flow over the telemetry port on the intel aero drone.
3. Visualize the local_position data on QGroundControl
4. Initially, the local position data gives good data then the periodic cut-off starts after 10-30 secs or so.

**Screenshots**
[image]

**Logs**
There is another issue on the intel drone that I am not able to get the flight logs. I have started a discussion here. When I find the solution to the logging problem I will upload the logs.
https://discuss.px4.io/t/unable-to-log-flight-data-on-intel-aero-with-px4-v1-9-2/15102

I understand this issue will be difficult to solve without the logs but the periodic pattern seems to be very distinctive and I was hoping someone could provide some suggestions to start debugging this issue.

Another point which I noted was the estimator_status.vel_ratio was 0 on the mavlink inspector.

Ashwin

julianoes, 2020-02-12:
This looks indeed very wrong. Could it be that the flow orientation is incorrect or something trivial like that?

ashwinsushil, 2020-02-12:
Hey, @julianoes  I usually use an IRIS+ drone for R&D. Checked the mounting/physical-orientation on the IRIS+ and the SENS_FLOW_ROT is set to Yaw 270(default) which seems to be working well. The intel drone (which is having this issue) is set up exactly the same way. So I don't think it is the orientation. 

Additional info: I am getting good data on the OPTICAL_FLOW_RAD.integrated_x and OPTICAL_FLOW_RAD.integrated_y when this phenomenon occurs.

For those who are interested to know about the mounting of PX4flow [here](https://docs.px4.io/v1.9.0/en/sensor/px4flow.html#mountingorientation) is the link

julianoes, 2020-02-14:
I would suggest to use master because some improvements for flying with Flow have been merged but I'm not sure if master still works on Intel Aero. The problem is that Intel Aero is not supported anymore by Intel.

closed 2020-05-15 by julianoes