PX4/PX4-Autopilot #14141: GPS & baro fusion problem

matejbone, 2020-02-12:
Hi!

I have a question regarding EKF2 fusion.

PX4 is set up to use GPS as a primary source for EKF2_HGT_MODE.
Reported altitude is OK until the GPS receiver is jammed. In that case, fusion creates a problem, because it merges invalid position from GPS and baro and from that point on fused altitude is incorrect. Px4 thinks it's flying way higher than it actually is.

Is there a way in the parameters to set up in case of invalid GPS to use baro as a primary source. Could I set PX4 to turn off fusion when GPS signal is invalid? I think that is feeding info in the Kalman filter even when it's invalid and from that point on fused altitude does not return to GPS reported one. 
Flight1
[image]

Flight2
I tried using EKF2_GPS_V_GATE and EKF2_GPS_P_GATE more restrictively but without any luck. 
[image]

Thanks!

julianoes, 2020-02-14:
Good question. Let me ping @bresch and @CarlOlsson hoping that they have answers.

CarlOlsson, 2020-02-14:
Unfortunately I don't have experience with GPS as primary height source. @matejbone Just curious, why don't you want to use baro as primary height source?

matejbone, 2020-02-14:
I tried to use GPS, because after a long flight 2+ hours I had in mind that barometric pressure would change and that would affect the approaching height. 

How do I adjust weight between both? Is there a parameter?

CarlOlsson, 2020-02-14:
> How do I adjust weight between both? Is there a parameter?

No the EKF only uses one of them at the time. Here is some super old issue regarding  baro offset estimation https://github.com/PX4/ecl/issues/44

julianoes, 2020-02-14:
FYI @RomanBapst @jkflying

jkflying, 2020-02-14:
Use baro as primary height source. The offset will be re-estimated using GPS if available, so you shouldn't have drift.

There was some talk about doing multiple height source fusion (vision, lidar, baro & GPS all together) but this would require a big rework of the EKF.

matejbone, 2020-02-17:
OK thanks for the help to all!

I will try to use baro and see what happens if GPS is jammed.

closed 2020-02-17 by matejbone

RomanBapst, 2020-02-17:
> Use baro as primary height source. The offset will be re-estimated using GPS if available, so you shouldn't have drift.
@jkflying Can you remind me where we are doing this? I was under the impression that we don't compensate for baro drift.

jkflying, 2020-02-17:
Hmm, it seems we only do that if we _don't_ use baro as primary.

https://github.com/PX4/ecl/blob/81c6d6655f6e409cac549ed8537b22996b413028/EKF/control.cpp#L1083-L1092