PX4/PX4-Autopilot #14142: Remove Mixer from PX4IO; Use Only Direct PWM Passthrough

JacobCrabill, 2020-02-12:
(Placeholder for discussion on this topic)

**Describe problem solved by the proposed feature**
The PX4IO processor is running out of memory (particularly for large mixer files)

**Describe your preferred solution**
A possible solution is to remove the mixer from px4iofirmware, and instead use direct pwm control from the FMU, rather than sending actuator controls. 

An example of this direct pwm control can be seen in the 'test' portion of the 'pwm' systemcmd.

**Additional context**
Other benefits of this change include simpler debugging of PWM outputs, as all mixing input/output would be available for viewing on the FMU.

Also worth noting is that this method is used by ArduPilot, which further leverages this by allowing the mixer mapping to be defined via run-time parameters, rather than baked-in mixer files.

mrpollo, 2020-02-12:
📌Subscribed

dlwalter, 2020-02-12:
This would be fantastic.  Currently using 8 simple mixers as a passthrough from an FMU mixer but there's all sorts of weirdness with NANs indicating disarmed that need to be handled and passed correctly.

jlecoeur, 2020-02-12:
Related PR: https://github.com/PX4/Firmware/pull/13581
This adds a pass-through mixer, with the short term goal to make it possible to do mixing either on both FMU and IO, and longer term goal to more features from IO to FMU.

bkueng, 2022-10-25:
Implemented

closed 2022-10-25 by bkueng

siddux, 2022-11-08:
> Implemented

Hi, has it been merged into main branch? If not in which branch can I found it?

Also, is there any documentation available?

Thanks.

bkueng, 2022-11-08:
Yes this is on `main`. What kind of documentation are you looking for?

siddux, 2022-11-08:
Yes, thanks I've found all I need...a long time without using PX4. Also, regarding documentation I think [this page](https://docs.px4.io/main/en/flight_stack/controller_diagrams.html) should be updated. It can be confusing with the new implementation using control allocation.

bkueng, 2022-11-09:
> Also, regarding documentation I think [this page](https://docs.px4.io/main/en/flight_stack/controller_diagrams.html) should be updated. It can be confusing with the new implementation using control allocation.

@hamishwillee apparently we missed one.