PX4/PX4-Autopilot #14146: Failure building ros2 workspace

Pman6543, 2020-02-12:
[image]

Whenever I try to build the ros1 and ros2 workspaces. The ros1 workspace succeeds, but the ros2 workspace fails. 

At first I tried 

source build_all.bash --ros1_ws_dir
The ros1 workspace would complete, but the ros2 build terminal would crash with no output, so I opted to try and build them manually. 

In calling this command 
colcon build --symlink-install --packages-skip ros1_bridge --event-handlers console_direct+
I get the above error.

julianoes, 2020-02-14:
I'm going to tag @TSC21 here. Hopefully he can help.

Pman6543, 2020-02-14:
@julianoes  I appreciate it, but I posted it in another directory where he has been helping me. I'm still encountering a great deal of issues.  But I'll continue to post there and close this issue. 

https://github.com/PX4/px4_ros_com/issues/37

closed 2020-02-14 by Pman6543