PX4/PX4-Autopilot #14149: Where to find quadrotor parameters and matrix allocation in the firmware ?

islamdib, 2020-02-13:
Hello,
Im new in this domaine. I want to modify the quadrotor PID parameters directly in hte firmware then to change the matrix allocation (control inputs U_i funtion of rotations speed Omega_i).

Please, enyone to tell me where to find this parameters ?

julianoes, 2020-02-14:
Can you point to the place in code where you want to change it?

islamdib, 2020-02-15:
I want to stabilize my quadrotor which is originaly in the X-morphology 
Ive four servomotors which allow to obtain an H-morphology (negociate in narrow space).
so the inertia matrix have to be changed then the allocation matrix too.
I dont know where to find this parameters ?

islamdib, 2020-02-19:
> Can you point to the place in code where you want to change it?

home/dib/Firmware/src/modules/mc_att_control/mc_att_control_params.c
Im not sure, but probably this is the corresponding folder

julianoes, 2020-02-20:
The controls just output angular rates and thrust. This is then mixed in a mixer. I think you want to read about that first: https://dev.px4.io/master/en/concept/mixing.html

saengphet, 2020-02-27:
> > Can you point to the place in code where you want to change it?
> 
> home/dib/Firmware/src/modules/mc_att_control/mc_att_control_params.c
> Im not sure, I think this is the corresponding folder

Are you looking for this?
https://github.com/PX4/Firmware/tree/master/src/lib/mixer/MultirotorMixer

saengphet, 2020-02-27:
> I want to stabilize my quadrotor which is originaly in the X-morphology
> Ive four servomotors which allow to obtain an H-morphology (negociate in narrow space).
> so the inertia matrix have to be changed then the allocation matrix too.
> I dont know where to find this parameters ?

Are you looking for this part?
https://github.com/PX4/Firmware/tree/master/src/lib/mixer/MultirotorMixer

islamdib, 2020-02-29:
@saengphet  
now Im working with simulink
the problem is that rotors haven't the same speed so I think that I have to change PID parameters in the first part.
In the second part , I didn't understand the command calculation formula:
% Quad X
 M1 = uint16(((Roll + Pitch - Yaw) / 2 * Thrust + Thrust) * single(1000)) + idle_PWM;
 M2 = uint16(((-Roll + Pitch + Yaw) / 2 * Thrust + Thrust) * single(1000)) + idle_PWM;
 M3 = uint16(((-Roll - Pitch - Yaw) / 2 * Thrust + Thrust) * single(1000)) + idle_PWM;
 M4 = uint16(((Roll - Pitch + Yaw) / 2 * Thrust + Thrust) * single(1000)) + idle_PWM;
 M5 = uint16(1000);
 M6 = uint16(1000);

[image]