PX4/PX4-Autopilot #14158: Transition Denied: ARMED to INIT while midflight

mickeyhorn, 2020-02-13:
**Describe the bug**
Seemingly randomly while flying, QGC would print out the following warning message:
"TRANSITION_DENIED: ARMED to INIT"
Nothing besides the warning would happen. QGC would just yell it out and the drone continued on its merry way.

**To Reproduce**
1) Drone connected to laptop running QGC via USB.
2) Mission file uploaded to the drone. 
[Chir_Test_2.1.20_medpoints_sortie0.txt](https://github.com/PX4/Firmware/files/4201399/Chir_Test_2.1.20_medpoints_sortie0.txt)
3) Drone disconnected from laptop. Powered on with LiPo battery.
4) Mission started. Drone flies normally.
5) Warning randomly pops up a few times midflight, but no change in drone behavior.

**Expected behavior**
I expect no warning to appear.

**Log Files and Screenshots**
https://review.px4.io/plot_app?log=80aa1e55-af2c-4483-bb2e-b60055cb08fb
You can see the warning message at the very bottom of this page.

**Drone (please complete the following information):**
It's a custom-built hexarotor using a Tarot 690S frame, a Pixhawk 2.1, the DJI E800 propulsion system, a Here+ RTK GPS, a FrSky X8R receiver, a Holybro 915MHz telemetry radio, a Mauch 4-14S BEC, and a Mauch current sensor.

**Additional context**
I am still running PX4 version 1.9.2 on this drone. It has flown multiple times before without receiving this warning, and has received minimal changes since it's last flight without the warning. The only change I can remember offhand is replacing the SD card in the Pixhawk, because we were having errors with our logs being cut short.

julianoes, 2020-02-14:
Nice pattern you flew there :smile:.

My only guess is that somehow you sent a calibration command to the vehicle, and you would then end up here:

https://github.com/PX4/Firmware/blob/342e0da7961f9a3301706ed3835cfc163b14b2ed/src/modules/commander/Commander.cpp#L3266-L3279

Was any other ground station connected? Or anything like MAVSDK or MAVROS sending commands?

mickeyhorn, 2020-02-18:
Hm, I don't believe we tried to start a calibration of any sort during the flight. And no, we only had the one ground station and 1 Taranis controller communicating with the drone.

julianoes, 2020-02-20:
I'm very confused then. Is there any other hint what went different on that flight? Any other observations?

mickeyhorn, 2020-02-20:
Not that I can remember. I'll try to push for our group to do another flight this weekend to try and replicate it.

smnogar, 2020-02-24:
I just had this happen, also running 1.9.2 (although slightly modified). We were sending offboard commands to the flight controller from MAVROS trying to test RC failsafe behavior. We had someone walk away from the vehicle holding the RC controller. When the RSSI signal went to zero, we got the error `Transition Denied: ARMED to INIT` instead of the RC failsafe triggering. The motors continued to spin. However, we have modified the state machine such that if RC is lost while receiving offboard commands, the RC failsafe triggers.

We repeated this test several times after this failure, and it worked as intended (i.e. the motors stopped spinning). So I'm not sure what was different about this test that we got this error.

Hardware was a pixracer and radios were spektrum.

smnogar, 2020-02-27:
On further investigation, I don't think this error caused the RC failsafe behavior. I think it was coincidental and that RC was never lost.

MaEtUgR, 2020-03-30:
> that somehow you sent a calibration command

It's enough to switch to the radio tab in QGC to send out such a command and trigger this warning.

MaEtUgR, 2020-03-30:
It's reproducable in SITL by opening the Radio tab see https://github.com/PX4/Firmware/issues/14527

dagar, 2020-05-04:
Should be fixed in https://github.com/PX4/Firmware/pull/14825.

dagar, 2020-05-06:
Fixed in #14825.

closed 2020-05-06 by dagar