PX4/PX4-Autopilot #14160: [sitl]px4_sitl gazebo_plane compile error

jinchengde, 2020-02-14:
**Describe the bug*
commander:
make distclean
make px4_sitl gazebo_plane
error:
[58/96] Building CXX object CMakeFiles...c/gazebo_geotagged_images_plugin.cpp.o
FAILED: CMakeFiles/gazebo_geotagged_images_plugin.dir/src/gazebo_geotagged_images_plugin.cpp.o 
/usr/bin/c++  -DLIBBULLET_VERSION=2.87 -DLIBBULLET_VERSION_GT_282 -Dgazebo_geotagged_images_plugin_EXPORTS -isystem /usr/include/gazebo-9 -isystem /usr/include/bullet -isystem /usr/include/simbody -isystem /usr/include/sdformat-6.0 -isystem /usr/include/ignition/math4 -isystem /usr/include/OGRE -isystem /usr/include/OGRE/Terrain -isystem /usr/include/OGRE/Paging -isystem /usr/include/ignition/transport4 -isystem /usr/include/ignition/msgs1 -isystem /usr/include/ignition/common1 -isystem /usr/include/ignition/fuel_tools1 -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -isystem /usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -I/home/jinchengde/GitHub/PX4/Firmware/Tools/sitl_gazebo/include -I. -I/usr/include/eigen3 -I/usr/include/eigen3/eigen3 -I/usr/include/gazebo-9/gazebo/msgs -I/home/jinchengde/GitHub/PX4/Firmware/mavlink/include -isystem /usr/include/opencv -I/home/jinchengde/GitHub/PX4/Firmware/Tools/sitl_gazebo/external/OpticalFlow/include -I/usr/include/gstreamer-1.0 -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -isystem /usr/include/uuid -isystem /usr/include/x86_64-linux-gnu -Wno-deprecated-declarations -fPIC   -I/usr/include/uuid -I/usr/include/x86_64-linux-gnu -std=gnu++1z -MD -MT CMakeFiles/gazebo_geotagged_images_plugin.dir/src/gazebo_geotagged_images_plugin.cpp.o -MF CMakeFiles/gazebo_geotagged_images_plugin.dir/src/gazebo_geotagged_images_plugin.cpp.o.d -o CMakeFiles/gazebo_geotagged_images_plugin.dir/src/gazebo_geotagged_images_plugin.cpp.o -c /home/jinchengde/GitHub/PX4/Firmware/Tools/sitl_gazebo/src/gazebo_geotagged_images_plugin.cpp
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
[63/96] Building CXX object CMakeFiles...dir/src/gazebo_mavlink_interface.cpp.o
ninja: build stopped: subcommand failed.
[714/718] Linking CXX executable bin/px4
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-build 
cd /home/jinchengde/GitHub/PX4/Firmware/build/px4_sitl_default/build_gazebo && /usr/bin/cmake --build .
ninja: build stopped: subcommand failed.
Makefile:205: recipe for target 'px4_sitl' failed
make: *** [px4_sitl] Error 1

julianoes, 2020-02-14:
Can you find the file like this?

```
find . -name SITLGps.pb.h
./build/px4_sitl_default/build_gazebo/SITLGps.pb.h
```

Also: please put ``` around pasted output.

jinchengde, 2020-02-15:
yes, I find the file in ./build/px4_sitl_default/build_gazebo/SITLGps.pb.h

and then I try ```make px4_sitl gazebo_plane``` again

this issue solved

julianoes, 2020-02-20:
So in a second attempt it works? In that case it sounds like a race because a dependency is not correctly specified for cmake.

julianoes, 2020-02-20:
I looked at the cmake file and I could not immediately see something but you can try this patch for Tools/sitl_gazebo:

```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index e79fb82..bc69b4a 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -296,11 +296,14 @@ PROTOBUF_GENERATE_CPP(PHY_PROTO_SRCS PHY_PROTO_HDRS ${physics_msgs})
 PROTOBUF_GENERATE_CPP(STD_PROTO_SRCS STD_PROTO_HDRS ${std_msgs})
 PROTOBUF_GENERATE_CPP(SEN_PROTO_SRCS SEN_PROTO_HDRS ${sensor_msgs})
 
+add_custom_target(sensor_msgs_headers ${SEN_PROTO_HDRS})
+
 add_library(mav_msgs SHARED ${MAV_PROTO_SRCS})
 add_library(nav_msgs SHARED ${NAV_PROTO_SRCS})
 add_library(physics_msgs SHARED ${PHY_PROTO_SRCS})
 add_library(std_msgs SHARED ${STD_PROTO_SRCS})
 add_library(sensor_msgs SHARED ${SEN_PROTO_SRCS})
+add_dependencies(sensor_msgs sensor_msgs_headers)
 
 #---------#
 # Plugins #
```

Would be intereseting to see if that resolves it.

jinchengde, 2020-02-22:
I had try but it not work

julianoes, 2020-02-26:
Ok, still the same error?

jinchengde, 2020-02-27:
yes, same error.

julianoes, 2020-03-01:
Do you know why this happens?

```
 c++: internal compiler error: Killed (program cc1plus)
```

Usually this happens if there is not enough RAM available. One thing to try is to reboot the computer and watch the free RAM during compilation using `top` or `htop`.

jinchengde, 2020-03-02:
you are correct, the computer I compile PX4 source code only have 4G memory, and used 
''' free -m ''' to monitor the memory usage, it looks all memory had be used up

julianoes, 2020-03-02:
Ok, in that case try with more RAM (or a swap partition) or make sure only one core is used.

closed 2020-03-02 by julianoes

shu-bamma, 2020-10-17:
> Ok, in that case try with more RAM (or a swap partition) or make sure only one core is used.

i am not on a virtual machine, still i am getting the exavt error while making this.
```
~/src/Firmware$make px4_sitl gazebo
-- PX4 version: v1.11.0-rc3-504-g7d6f8dc
-- PX4 config file: /home/shubham/src/Firmware/boards/px4/sitl/default.cmake
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is GNU 5.4.0
-- The C compiler identification is GNU 5.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working CXX compiler using: Ninja
-- Check for working CXX compiler using: Ninja -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler using: Ninja
-- Check for working C compiler using: Ninja -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Building for code coverage
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.12", minimum required is "3") 
-- build type is RelWithDebInfo
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-456-gf666ebb
-- ROMFS: px4fmu_common
-- Configuring done
-- Generating done
-- Build files have been written to: /home/shubham/src/Firmware/build/px4_sitl_default
[2/767] git submodule src/lib/ecl
[11/767] git submodule src/drivers/gps/devices
[12/767] git submodule mavlink/include/mavlink/v2.0
[15/767] git submodule Tools/sitl_gazebo
[20/767] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- cmake build type: RelWithDebInfo
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler using: Ninja
-- Check for working C compiler using: Ninja -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler using: Ninja
-- Check for working CXX compiler using: Ninja -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test COMPILER_SUPPORTS_CXX17
-- Performing Test COMPILER_SUPPORTS_CXX17 - Success
-- Performing Test COMPILER_SUPPORTS_CXX14
-- Performing Test COMPILER_SUPPORTS_CXX14 - Success
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Using C++14 standard
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
[... 93 lines cut ...]
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
[58/112] Building CXX object CMakeFile...dir/src/gazebo_gst_camera_plugin.cpp.o
FAILED: /usr/bin/c++   -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NO_DEBUG -Dgazebo_gst_camera_plugin_EXPORTS -isystem /usr/include/qt4 -isystem /usr/include/qt4/QtGui -isystem /usr/include/qt4/QtCore -isystem /usr/include/gazebo-7 -isystem /usr/include/sdformat-4.4 -isystem /usr/include/ignition/math2 -isystem /usr/include/OGRE -isystem /usr/include/OGRE/Terrain -isystem /usr/include/OGRE/Paging -I/home/shubham/src/Firmware/Tools/sitl_gazebo/include -I. -I/usr/include/eigen3 -I/usr/include/eigen3/eigen3 -I/usr/include/gazebo-7/gazebo/msgs -I/home/shubham/src/Firmware/mavlink/include -isystem /opt/ros/kinetic/include/opencv-3.3.1-dev -isystem /opt/ros/kinetic/include/opencv-3.3.1-dev/opencv -I/home/shubham/src/Firmware/Tools/sitl_gazebo/external/OpticalFlow/include -I/usr/include/gstreamer-1.0 -I/usr/lib/x86_64-linux-gnu/gstreamer-1.0/include -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -O2 -g -DNDEBUG -fPIC   -Wno-deprecated-declarations -Wno-address-of-packed-member -std=gnu++14 -MMD -MT CMakeFiles/gazebo_gst_camera_plugin.dir/src/gazebo_gst_camera_plugin.cpp.o -MF CMakeFiles/gazebo_gst_camera_plugin.dir/src/gazebo_gst_camera_plugin.cpp.o.d -o CMakeFiles/gazebo_gst_camera_plugin.dir/src/gazebo_gst_camera_plugin.cpp.o -c /home/shubham/src/Firmware/Tools/sitl_gazebo/src/gazebo_gst_camera_plugin.cpp
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
[58/112] Building CXX object CMakeFile...c/gazebo_geotagged_images_plugin.cpp.o
FAILED: /usr/bin/c++   -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NO_DEBUG -Dgazebo_geotagged_images_plugin_EXPORTS -isystem /usr/include/qt4 -isystem /usr/include/qt4/QtGui -isystem /usr/include/qt4/QtCore -isystem /usr/include/gazebo-7 -isystem /usr/include/sdformat-4.4 -isystem /usr/include/ignition/math2 -isystem /usr/include/OGRE -isystem /usr/include/OGRE/Terrain -isystem /usr/include/OGRE/Paging -I/home/shubham/src/Firmware/Tools/sitl_gazebo/include -I. -I/usr/include/eigen3 -I/usr/include/eigen3/eigen3 -I/usr/include/gazebo-7/gazebo/msgs -I/home/shubham/src/Firmware/mavlink/include -isystem /opt/ros/kinetic/include/opencv-3.3.1-dev -isystem /opt/ros/kinetic/include/op [... cut ...]
See <file:///usr/share/doc/gcc-5/README.Bugs> for instructions.
[58/112] Building CXX object CMakeFile...plugin.dir/src/gazebo_imu_plugin.cpp.o
ninja: build stopped: subcommand failed.
[764/767] Linking CXX executable bin/px4
FAILED: cd /home/shubham/src/Firmware/build/px4_sitl_default/build_gazebo && /usr/bin/cmake --build /home/shubham/src/Firmware/build/px4_sitl_default/build_gazebo
ninja: build stopped: subcommand failed.
Makefile:224: recipe for target 'px4_sitl' failed
make: *** [px4_sitl] Error 1

```
@julianoes please help