PX4/PX4-Autopilot #14163: Unable to run Gazebo

Pman6543, 2020-02-14:
When I try to run 

sudo make px4_sitl_default gazebo this happens

[0/5] Performing build step for 'sitl_gazebo'
[0/1] Re-running CMake...
-- install-prefix: /usr/local
-- Using C++17 compiler
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   system
--   thread
--   filesystem
--   chrono
--   date_time
--   atomic
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   thread
--   system
--   filesystem
--   program_options
--   regex
--   iostreams
--   date_time
--   chrono
--   atomic
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread (found version "3.0.0") 
-- Boost version: 1.65.1
-- Looking for OGRE...
-- Found Ogre Ghadamon (1.9.0)
-- Looking for OGRE_Paging...
-- Found OGRE_Paging: optimized;/usr/lib/x86_64-linux-gnu/libOgrePaging.so;debug;/usr/lib/x86_64-linux-gnu/libOgrePaging.so
-- Looking for OGRE_Terrain...
-- Found OGRE_Terrain: optimized;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so
-- Looking for OGRE_Property...
-- Found OGRE_Property: optimized;/usr/lib/x86_64-linux-gnu/libOgreProperty.so;debug;/usr/lib/x86_64-linux-gnu/libOgreProperty.so
-- Looking for OGRE_RTShaderSystem...
-- Found OGRE_RTShaderSystem: optimized;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so;debug;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so
-- Looking for OGRE_Volume...
-- Found OGRE_Volume: optimized;/usr/lib/x86_64-linux-gnu/libOgreVolume.so;debug;/usr/lib/x86_64-linux-gnu/libOgreVolume.so
-- Looking for OGRE_Overlay...
-- Found OGRE_Overlay: optimized;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so;debug;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread;-lpthread (found suitable version "3.0.0", minimum required is "2.3.0") 
-- Config-file not installed for ZeroMQ -- checking for pkg-config
-- Checking for module 'libzmq >= 4'
--   Found libzmq , version 4.2.5
-- Checking for module 'uuid'
--   Found uuid, version 2.31.1
-- Checking for module 'tinyxml2'
--   Found tinyxml2, version 6.0.0
-- Looking for dlfcn.h - found
-- Looking for libdl - found
-- FreeImage.pc not found, we will search for FreeImage_INCLUDE_DIRS and FreeImage_LIBRARIES
-- Checking for module 'gts'
--   Found gts, version 0.7.6
-- Checking for module 'libswscale'
--   Found libswscale, version 4.8.100
-- Checking for module 'libavdevice >= 56.4.100'
--   Found libavdevice , version 57.10.100
-- Checking for module 'libavformat'
--   Found libavformat, version 57.83.100
-- Checking for module 'libavcodec'
--   Found libavcodec, version 57.107.100
-- Checking for module 'libavutil'
--   Found libavutil, version 55.78.100
-- Checking for module 'jsoncpp'
--   Found jsoncpp, version 1.7.4
-- Checking for module 'yaml-0.1'
--   Found yaml-0.1, version 0.1.7
-- Checking for module 'libzip'
--   Found libzip, version 1.1.2
CMake Error at CMakeLists.txt:76 (add_subdirectory):
  The source directory

    /home/nrivera/src/Firmware/Tools/sitl_gazebo/external/OpticalFlow

  does not contain a CMakeLists.txt file.

-- catkin DISABLED
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread;-lpthread;-lpthread (found version "3.0.0") 
-- Gazebo version: 9.12
-- Found GStreamer: adding gst_camera_plugin
-- Found GStreamer: adding gst_video_stream_widget
-- Configuring incomplete, errors occurred!
See also "/home/nrivera/src/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeOutput.log".
See also "/home/nrivera/src/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeError.log".
FAILED: build.ninja 
/usr/bin/cmake -H/home/nrivera/src/Firmware/Tools/sitl_gazebo -B/home/nrivera/src/Firmware/build/px4_sitl_default/build_gazebo
ninja: error: rebuilding 'build.ninja': subcommand failed
[1/5] Generating ../../logs
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-build 
cd /home/nrivera/src/Firmware/build/px4_sitl_default/build_gazebo && /usr/bin/cmake --build .
ninja: build stopped: subcommand failed.
Makefile:198: recipe for target 'px4_sitl_default' failed
make: *** [px4_sitl_default] Error 1

What is going on? Also when I try to do

make px4_sitl_default gazebo 

I get this

ninja: error: opening build log: Permission denied
Makefile:198: recipe for target 'px4_sitl_default' failed
make: *** [px4_sitl_default] Error 1

TSC21, 2020-02-14:
Use `git submodule update --init --recursive`.

closed 2020-02-14 by TSC21

Pman6543, 2020-02-14:
```
nrivera@ubuntu:~/src/Firmware$ make px4_sitl gazebo
[0/5] Performing build step for 'sitl_gazebo'
ninja: error: opening build log: Permission denied
[1/5] Generating ../../logs
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-build 
cd /home/nrivera/src/Firmware/build/px4_sitl_default/build_gazebo && /usr/bin/cmake --build .
ninja: build stopped: subcommand failed.
Makefile:205: recipe for target 'px4_sitl' failed
make: *** [px4_sitl] Error 1
nrivera@ubuntu:~/src/Firmware$ sudo make px4_sitl gazebo
[0/5] Performing build step for 'sitl_gazebo'
[1/60] Building CXX object CMakeFiles/...c/gazebo_geotagged_images_plugin.cpp.o
FAILED: CMakeFiles/gazebo_geotagged_images_plugin.dir/src/gazebo_geotagged_images_plugin.cpp.o 
/usr/bin/c++  -DLIBBULLET_VERSION=2.87 -DLIBBULLET_VERSION_GT_282 -Dgazebo_geotagged_images_plugin_EXPORTS -isystem /usr/include/gazebo-9 -isystem /usr/include/bullet -isystem /usr/include/simbody -isystem /usr/include/sdformat-6.2 -isystem /usr/include/ignition/math4 -isystem /usr/include/OGRE -isystem /usr/include/OGRE/Terrain -isystem /usr/include/OGRE/Paging -isystem /usr/include/ignition/transport4 -isystem /usr/include/ignition/msgs1 -isystem /usr/include/ignition/common1 -isystem /usr/include/ignition/fuel_tools1 -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -isystem /usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -I/home/nrivera/src/Firmware/Tools/sitl_gazebo/include -I. -I/usr/include/eigen3 -I/usr/include/eigen3/eigen3 -I/usr/include/gazebo-9/gazebo/msgs -I/home/nrivera/src/Firmware/mavlink/include -isystem /usr/include/opencv -I/home/nrivera/src/Firmware/Tools/sitl_gazebo/external/OpticalFlow/include -I/usr/include/gstreamer-1.0 -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -isystem /usr/include/uuid -isystem /usr/include/x86_64-linux-gnu -Wno-deprecated-declarations -fPIC   -I/usr/include/uuid -I/usr/include/x86_64-linux-gnu -std=gnu++1z -MD -MT CMakeFiles/gazebo_geotagged_images_plugin.dir/src/gazebo_geotagged_images_plugin.cpp.o -MF CMakeFiles/gazebo_geotagged_images_plugin.dir/src/gazebo_geotagged_images_plugin.cpp.o.d -o CMakeFiles/gazebo_geotagged_images_plugin.dir/src/gazebo_geotagged_images_plugin.cpp.o -c /home/nrivera/src/Firmware/Tools/sitl_gazebo/src/gazebo_geotagged_images_plugin.cpp
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
[2/60] Building CXX object CMakeFiles/...ugin.dir/src/gazebo_lidar_plugin.cpp.o
FAILED: CMakeFiles/gazebo_lidar_plugin.dir/src/gazebo_lidar_plugin.cpp.o 
/usr/bin/c++  -DLIBBULLET_VERSION=2.87 -DLIBBULLET_VERSION_GT_282 -Dgazebo_lidar_plugin_EXPORTS -isystem /usr/include/gazebo-9 -isystem /usr/include/bullet -isystem /usr/include/simbody -isystem /usr/include/sdformat-6.2 -isystem /usr/include/ignition/math4 -isystem /usr/include/OGRE -isystem /usr/include/OGRE/Terrain -isystem /usr/include/OGRE/Paging -isystem /usr/include/ignition/transport4 -isystem /usr/include/ignition/msgs1 -isystem /usr/include/ignition/common1 -isystem /usr/include/ignition/fuel_tools1 -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -isystem /usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -I/home/nrivera/src/Firmware/Tools/sitl_gazebo/include -I. -I/usr/include/eigen3 -I/usr/include/eigen3/eigen3 -I/usr/include/gazebo-9/gazebo/msgs -I/home/nrivera/src/Firmware/mavlink/include -isystem /usr/include/opencv -I/home/nrivera/src/Firmware/Tools/sitl_gazebo/external/OpticalFlow/include -I/usr/include/gstreamer-1.0 -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -isystem /usr/include/uuid -isystem /usr/include/x86_64-linux-gnu -Wno-deprecated-declarations -fPIC   -I/usr/include/uuid -I/usr/include/x86_64-linux-gnu -std=gnu++1z -MD -MT CMakeFiles/gazebo_lidar_plugin.dir/src/gazebo_lidar_plugin.cpp.o -MF CMakeFiles/gazebo_lidar_plugin.dir/src/gazebo_lidar_plugin.cpp.o.d -o CMakeFiles/gazebo_lidar_plugin.dir/src/gazebo_lidar_plugin.cpp.o -c /home/nrivera/src/Firmware/Tools/sitl_gazebo/src/gazebo_lidar_plugin.cpp
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
[3/60] Building CXX object CMakeFiles/...gin.dir/src/gazebo_vision_plugin.cpp.o
FAILED: CMakeFiles/gazebo_vision_plugin.dir/src/gazebo_vision_plugin.cpp.o 
/usr/bin/c++  -DLIBBULLET_VERSION=2.87 -DLIBBULLET_VERSION_GT_282 -Dgazebo_vision_plugin_EXPORTS -isystem /usr/include/gazebo-9 -isystem /usr/include/bullet -isystem /usr/include/simbody -isystem /usr/include/sdformat-6.2 -isystem /usr/include/ignition/math4 -isystem /usr/include/OGRE -isystem /usr/include/OGRE/Terrain -isystem /usr/include/OGRE/Paging -isystem /usr/include/ignition/transport4 -isystem /usr/include/ignition/msgs1 -isystem /usr/include/ignition/common1 -isystem /usr/include/ignition/fuel_tools1 -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -isystem /usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -I/home/nrivera/src/Firmware/Tools/sitl_gazebo/include -I. -I/usr/include/eigen3 -I/usr/include/eigen3/eigen3 -I/usr/include/gazebo-9/gazebo/msgs -I/home/nrivera/src/Firmware/mavlink/include -isystem /usr/include/opencv -I/home/nrivera/src/Firmware/Tools/sitl_gazebo/external/OpticalFlow/incl [... cut ...]
[6/60] Building CXX object CMakeFiles/...plugin.dir/src/gazebo_uuv_plugin.cpp.o
ninja: build stopped: subcommand failed.
[1/5] Generating ../../logs
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-build 
cd /home/nrivera/src/Firmware/build/px4_sitl_default/build_gazebo && /usr/bin/cmake --build .
ninja: build stopped: subcommand failed.
Makefile:205: recipe for target 'px4_sitl' failed
make: *** [px4_sitl] Error 1
nrivera@ubuntu:~/src/Firmware$ 

```

This was my result after doing 

git submodule update --init --recursive.

Pman6543, 2020-02-14:
I deleted the entire Firmware directory and re-cloned it. 

I'm still getting the error.

```
nrivera@ubuntu:~/src/Firmware$ sudo make px4_sitl gazebo_typhoon_h480
[sudo] password for nrivera: 
[139/654] Performing build step for 'sitl_gazebo'
[3/96] Generating /home/nrivera/src/Firmware/Tools/sitl_gazebo/models/3DR_gps_mag/3DR_gps_mag-gen.sdf
/home/nrivera/src/Firmware/Tools/sitl_gazebo/models/3DR_gps_mag/3DR_gps_mag.sdf.jinja -> /home/nrivera/src/Firmware/Tools/sitl_gazebo/models/3DR_gps_mag/3DR_gps_mag-gen.sdf
[4/96] Generating /home/nrivera/src/Firmware/Tools/sitl_gazebo/models/mb1240-xl-ez4/mb1240-xl-ez4-gen.sdf
/home/nrivera/src/Firmware/Tools/sitl_gazebo/models/mb1240-xl-ez4/mb1240-xl-ez4.sdf.jinja -> /home/nrivera/src/Firmware/Tools/sitl_gazebo/models/mb1240-xl-ez4/mb1240-xl-ez4-gen.sdf
[5/96] Generating /home/nrivera/src/Firmware/Tools/sitl_gazebo/models/c920/c920-gen.sdf
/home/nrivera/src/Firmware/Tools/sitl_gazebo/models/c920/c920.sdf.jinja -> /home/nrivera/src/Firmware/Tools/sitl_gazebo/models/c920/c920-gen.sdf
[7/96] Generating /home/nrivera/src/Firmware/Tools/sitl_gazebo/models/matrice_100/matrice_100-gen.sdf
/home/nrivera/src/Firmware/Tools/sitl_gazebo/models/matrice_100/matrice_100.sdf.jinja -> /home/nrivera/src/Firmware/Tools/sitl_gazebo/models/matrice_100/matrice_100-gen.sdf
[9/96] Generating /home/nrivera/src/Firmware/Tools/sitl_gazebo/models/pixhawk/pixhawk-gen.sdf
/home/nrivera/src/Firmware/Tools/sitl_gazebo/models/pixhawk/pixhawk.sdf.jinja -> /home/nrivera/src/Firmware/Tools/sitl_gazebo/models/pixhawk/pixhawk-gen.sdf
[10/96] Generating /home/nrivera/src/Firmware/Tools/sitl_gazebo/models/px4flow/px4flow-gen.sdf
/home/nrivera/src/Firmware/Tools/sitl_gazebo/models/px4flow/px4flow.sdf.jinja -> /home/nrivera/src/Firmware/Tools/sitl_gazebo/models/px4flow/px4flow-gen.sdf
[11/96] Generating /home/nrivera/src/Firmware/Tools/sitl_gazebo/models/r200/r200-gen.sdf
/home/nrivera/src/Firmware/Tools/sitl_gazebo/models/r200/r200.sdf.jinja -> /home/nrivera/src/Firmware/Tools/sitl_gazebo/models/r200/r200-gen.sdf
[12/96] Generating /home/nrivera/src/Firmware/Tools/sitl_gazebo/models/iris/iris.sdf

[24/96] Generating /home/nrivera/src/Firmware/Tools/sitl_gazebo/models/sf10a/sf10a-gen.sdf
/home/nrivera/src/Firmware/Tools/sitl_gazebo/models/sf10a/sf10a.sdf.jinja -> /home/nrivera/src/Firmware/Tools/sitl_gazebo/models/sf10a/sf10a-gen.sdf
[39/96] Building CXX object CMakeFiles/gazebo_uuv_plugin.dir/src/gazebo_uuv_plugin.cpp.o
FAILED: CMakeFiles/gazebo_uuv_plugin.dir/src/gazebo_uuv_plugin.cpp.o 
/usr/bin/c++  -DLIBBULLET_VERSION=2.87 -DLIBBULLET_VERSION_GT_282 -Dgazebo_uuv_plugin_EXPORTS -isystem /usr/include/gazebo-9 -isystem /usr/include/bullet -isystem /usr/include/simbody -isystem /usr/include/sdformat-6.2 -isystem /usr/include/ignition/math4 -isystem /usr/include/OGRE -isystem /usr/include/OGRE/Terrain -isystem /usr/include/OGRE/Paging -isystem /usr/include/ignition/transport4 -isystem /usr/include/ignition/msgs1 -isystem /usr/include/ignition/common1 -isystem /usr/include/ignition/fuel_tools1 -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -isystem /usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -I/home/nrivera/src/Firmware/Tools/sitl_gazebo/include -I. -I/usr/include/eigen3 -I/usr/include/eigen3/eigen3 -I/usr/include/gazebo-9/gazebo/msgs -I/home/nrivera/src/Firmware/mavlink/include -isystem /usr/include/opencv -I/home/nrivera/src/Firmware/Tools/sitl_gazebo/external/OpticalFlow/include -I/usr/include/gstreamer-1.0 -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -isystem /usr/include/uuid -isystem /usr/include/x86_64-linux-gnu -Wno-deprecated-declarations -fPIC   -I/usr/include/uuid -I/usr/include/x86_64-linux-gnu -std=gnu++1z -MD -MT CMakeFiles/gazebo_uuv_plugin.dir/src/gazebo_uuv_plugin.cpp.o -MF CMakeFiles/gazebo_uuv_plugin.dir/src/gazebo_uuv_plugin.cpp.o.d -o CMakeFiles/gazebo_uuv_plugin.dir/src/gazebo_uuv_plugin.cpp.o -c /home/nrivera/src/Firmware/Tools/sitl_gazebo/src/gazebo_uuv_plugin.cpp
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
[40/96] Building CXX object CMakeFiles/gazebo_lidar_plugin.dir/src/gazebo_lidar_plugin.cpp.o
FAILED: CMakeFiles/gazebo_lidar_plugin.dir/src/gazebo_lidar_plugin.cpp.o 
/usr/bin/c++  -DLIBBULLET_VERSION=2.87 -DLIBBULLET_VERSION_GT_282 -Dgazebo_lidar_plugin_EXPORTS -isystem /usr/include/gazebo-9 -isystem /usr/include/bullet -isystem /usr/include/simbody -isystem /usr/include/sdformat-6.2 -isystem /usr/include/ignition/math4 -isystem /usr/include/OGRE -isystem /usr/include/OGRE/Terrain -isystem /usr/include/OGRE/Paging -isystem /usr/include/ignition/transport4 -isystem /usr/include/ignition/msgs1 -isystem /usr/include/ignition/common1 -isystem /usr/include/ignition/fuel_tools1 -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -isystem /usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -I/home/nrivera/src/Firmware/Tools/sitl_gazebo/include -I. -I/usr/include/eigen3 -I/usr/include/eigen3/eigen3 -I/usr/include/gazebo-9/gazebo/msgs -I/home/nrivera/src/Firmware/mavlink/include -isystem /usr/include/opencv -I/home/nrivera/src/Firmware/Tools/sitl_gazebo/external/OpticalFlow/include -I/usr/include/gstreamer-1.0 -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -isystem /usr/include/uuid -isystem /usr/inclu [... cut ...]
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
[44/96] Building CXX object CMakeFiles/gazebo_geotagged_images_plugin.dir/src/gazebo_geotagged_images_plugin.cpp.o
ninja: build stopped: subcommand failed.
[650/654] Linking CXX executable bin/px4
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-build 
cd /home/nrivera/src/Firmware/build/px4_sitl_default/build_gazebo && /usr/bin/cmake --build .
ninja: build stopped: subcommand failed.
Makefile:205: recipe for target 'px4_sitl' failed
make: *** [px4_sitl] Error 1

```

Pman6543, 2020-02-18:
To any poor soul who ends up with the same error I had I found the solution. My virtual  machine did not have enough ram thus kept crashing the sim. 

Run dmesg and you should see if this was the case for you. 

My vm was failing at 4gb, so I brought it up to 7. It might work at 5 or 6, but I just wanted to be safe.

Duhan07, 2021-07-08:
I'm using dual boot but I'm getting same error.