PX4/PX4-Autopilot #14164: Error about px4 sitl_gazebo in gazebo

myboyhood, 2020-02-15:
# Hello everyone, I am new to px4,though sitl_gazebo can run successfully before, 
But I got this error now:
```
gzserver: symbol lookup error: /home/wzy/src/Firmware/build/px4_sitl_default/build_gazebo/libgazebo_multirotor_base_plugin.so: undefined symbol: _ZN8mav_msgs4msgs10MotorSpeedC1Ev
[gazebo-2] process has died [pid 17280, exit code 127, cmd /opt/ros/melodic/lib/gazebo_ros/gzserver --verbose -e ode /home/wzy/catkin_ws/src/my_gazebo_world/world/my_world.world __name:=gazebo __log:=/home/wzy/.ros/log/096051f0-4fe6-11ea-8760-b025aa2f887e/gazebo-2.log].
log file: /home/wzy/.ros/log/096051f0-4fe6-11ea-8760-b025aa2f887e/gazebo-2*.log
```
After I open gazebo ,it seems normal.

But when I launch my model , px4 and mavros in a launch file below, I got this error.

I am not edit libgazebo_multirotor_base_plugin relevant .cpp or .hpp

I see some help in [gazebo](https://bitbucket.org/osrf/gazebo/issues/2448/problem-running-gazebo7),  I even reinstall my ~/src/Firmware 
## But it is not work for me ,thanks for any help!!!
# my world launch
```
    <include file="$(find gazebo_ros)/launch/empty_world.launch">
      <arg name="world_name" default="$(find my_gazebo_world)/world/my_world.world"/>
      <arg name="paused" value="false"/>
      <arg name="use_sim_time" value="true"/>
      <arg name="gui" value="true"/>
      <arg name="headless" value="false"/>
      <arg name="debug" value="false"/>
      <arg name="verbose" default="true"/>
      <arg name="respawn_gazebo" default="false"/>
    </include>
```

# my model px4 mavros  launch
```
    <arg name="vehicle" default="iris"/>
    <!--<arg name="my_model" default="my_iris_fpv_lidar"/>-->
    <arg name="my_model" default="iris"/>
    <arg name="sdf" default="$(find mavlink_sitl_gazebo)/models/$(arg my_model)/$(arg my_model).sdf"/>
    <arg name="est" default="ekf2"/>
    <env name="PX4_SIM_MODEL" value="$(arg vehicle)" />
    <env name="PX4_ESTIMATOR" value="$(arg est)" />
    <node name="plane" pkg="gazebo_ros" type="spawn_model" output="screen" args="-sdf -file $(arg sdf) -model $(arg vehicle) -x 1 -y 1 -z 0 -R 0 -P 0 -Y 0"/>

    <arg name="fcu_url" default="udp://:14540@127.0.0.1:14557"/>
    <arg name="respawn_mavros" default="false"/>
    <!-- PX4 configs -->
    <arg name="interactive" default="true"/>
    <arg unless="$(arg interactive)" name="px4_command_arg1" value="-d"/>
    <arg     if="$(arg interactive)" name="px4_command_arg1" value=""/>
    <node name="sitl" pkg="px4" type="px4" output="screen"
          args="$(find px4)/ROMFS/px4fmu_common -s etc/init.d-posix/rcS $(arg px4_command_arg1)" required="true"/>

    <!-- MAVROS -->
    <!-- GCS link is provided by SITL -->
    <arg name="gcs_url" value=""/>
<!--    <arg name="fcu_url" value="$(arg fcu_url)"/>-->
    <arg name="tgt_system" default="1" />
    <arg name="tgt_component" default="1" />
    <arg name="log_output" default="screen" />
    <arg name="fcu_protocol" default="v2.0" />
    <!--<arg name="respawn_mavros" default="false" />-->
    <!--px4_config.yaml-->
    <arg name="pluginlists_yaml" value="$(find mavros)/launch/px4_pluginlists.yaml" />
    <arg name="config_yaml" value="$(find mavros)/launch/px4_config.yaml" />
    <!-- mavros node.launch-->
    <node pkg="mavros" type="mavros_node" name="mavros" required="$(eval not respawn_mavros)" clear_params="true" output="$(arg log_output)" respawn="$(arg respawn_mavros)">
        <param name="fcu_url" value="$(arg fcu_url)" />
        <param name="gcs_url" value="$(arg gcs_url)" />
        <param name="target_system_id" value="$(arg tgt_system)" />
        <param name="target_component_id" value="$(arg tgt_component)" />
        <param name="fcu_protocol" value="$(arg fcu_protocol)" />

        <!-- load blacklist, config -->
        <rosparam command="load" file="$(arg pluginlists_yaml)" />
        <rosparam command="load" file="$(arg config_yaml)" />
    </node>
```

Jaeyoung-Lim, 2020-02-17:
@myboyhood You need to set the model path properly. The instructions can be found here: http://dev.px4.io/master/en/simulation/ros_interface.html

myboyhood, 2020-02-17:
> @myboyhood You need to set the model path properly. The instructions can be found here: http://dev.px4.io/master/en/simulation/ros_interface.html

thanks,but I check my ~/.bashrc
```
export GAZEBO_MODEL_PATH=/usr/share/gazebo-9/models:/home/wzy/src/Firmware/Tools/sitl_gazebo/models:/home/wzy/.gazebo/models
```
and `env` also has above environment variable
I also check my other path
```
LD_LIBRARY_PATH=/home/wzy/mavros_ws/devel/lib:/home/wzy/catkin_ws/devel/lib:/opt/ros/melodic/lib:/home/wzy/src/Firmware/build/px4_sitl_default/build_gazebo:/usr/lib/x86_64-linux-gnu/gazebo-9/plugins
```
```
GAZEBO_PLUGIN_PATH=/usr/lib/x86_64-linux-gnu/gazebo-9/plugins::/home/wzy/src/Firmware/build/px4_sitl_default/build_gazebo:/home/wzy/src/Firmware/Tools/sitl_gazebo/build
```
I also have these in ~/.bashrc
```
source /home/wzy/src/Firmware/Tools/setup_gazebo.bash /home/wzy/src/Firmware /home/wzy/src/Firmware/build/px4_sitl_default
```
```
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:/home/wzy/src/Firmware:/home/wzy/src/Firmware/Tools/sitl_gazebo
```
Sorry ,I can not figure out my where is wrong about model path,thanks!!

Jaeyoung-Lim, 2020-02-17:
The `GAZEBO_PLUGIN_PATH` needs to point to the build directory of the firmware, and not the build directory inside sitl_gazebo
```
GAZEBO_PLUGIN_PATH=<Firmware_path>/Firmware/build/px4_sitl_default/build_gazebo
```

myboyhood, 2020-02-18:
> The `GAZEBO_PLUGIN_PATH` needs to point to the build directory of the firmware, and not the build directory inside sitl_gazebo
> 
> ```
> GAZEBO_PLUGIN_PATH=<Firmware_path>/Firmware/build/px4_sitl_default/build_gazebo
> ```

yes,I set the GAZEBO_PLUGIN_PATH as you said above, unluckily, it still report this error,
I see a similar [issue](https://github.com/PX4/Firmware/issues/10487)
so I reinstall gazebo by debian ,
and under `~/src/Firmware`, I run `make px4_sitl_default gazebo`
got the same error below :-(
```
wzy@wzy-lab:~/src/Firmware$ make px4_sitl_default gazebo
[0/4] Performing build step for 'sitl_gazebo'
ninja: no work to do.
[3/4] cd /home/wzy/src/Firmware/build/...zy/src/Firmware/build/px4_sitl_default
SITL ARGS
sitl_bin: /home/wzy/src/Firmware/build/px4_sitl_default/bin/px4
debugger: none
program: gazebo
model: none
src_path: /home/wzy/src/Firmware
build_path: /home/wzy/src/Firmware/build/px4_sitl_default
empty model, setting iris as default
Gazebo multi-robot simulator, version 9.12.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.1.103
gzserver: symbol lookup error: /home/wzy/src/Firmware/build/px4_sitl_default/build_gazebo/libgazebo_multirotor_base_plugin.so: undefined symbol: _ZN8mav_msgs4msgs10MotorSpeedC1Ev

```

julianoes, 2020-03-02:
@myboyhood have you tried to properly rebuild everything after a clean?

myboyhood, 2020-03-02:
> 
> 
> @myboyhood have you tried to properly rebuild everything after a clean?

yes, I reinstall gazebo and px4 firmware respectively, I test rotors simulation from ETH in gazebo, there is no error, I think there is some links between gazebo and px4_sitl not removed totally, the problem may be caused by my fool adaption of source cpp. but I reinstall px4 firmware cannot help to solve....

julianoes, 2020-03-02:
Check what dependencies `px4` has:

```
ldd /home/wzy/src/Firmware/build/px4_sitl_default/bin/px4
```

myboyhood, 2020-03-03:
**yes** , I try px4 and others like px4-motor_test , both of  them list below, sorry I cannot figure out whether this is wrong ??
```
wzy@wzy-lab:~$ ldd /home/wzy/src/Firmware/build/px4_sitl_default/bin/px4
	linux-vdso.so.1 (0x00007ffe9c93a000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f66a9d83000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f66a9b7b000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f66a9977000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f66a95ee000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f66a9250000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f66a8e5f000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f66aa244000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f66a8c47000)
```
I also call spawn_model in gazebo_ros pkg to load iris.sdf, it seem nothing wrong from the log below , but I find in gazebo ,the only half body of iris model is  above horizontal plane. I don't know what cause this weird phenomenon  :-(
```
[rospy.client][INFO] 2020-03-03 08:50:33,380: init_node, name[/plane], pid[5220]
[xmlrpc][INFO] 2020-03-03 08:50:33,380: XML-RPC server binding to 0.0.0.0:0
[xmlrpc][INFO] 2020-03-03 08:50:33,380: Started XML-RPC server [http://wzy-lab:43727/]
[rospy.init][INFO] 2020-03-03 08:50:33,380: ROS Slave URI: [http://wzy-lab:43727/]
[rospy.impl.masterslave][INFO] 2020-03-03 08:50:33,380: _ready: http://wzy-lab:43727/
[rospy.registration][INFO] 2020-03-03 08:50:33,381: Registering with master node http://localhost:11311
[xmlrpc][INFO] 2020-03-03 08:50:33,381: xml rpc node: starting XML-RPC server
[rospy.init][INFO] 2020-03-03 08:50:33,481: registered with master
[rospy.rosout][INFO] 2020-03-03 08:50:33,481: initializing /rosout core topic
[rospy.rosout][INFO] 2020-03-03 08:50:33,482: connected to core topic /rosout
[rospy.simtime][INFO] 2020-03-03 08:50:33,483: initializing /clock core topic
[rospy.simtime][INFO] 2020-03-03 08:50:33,485: connected to core topic /clock
[rosout][INFO] 2020-03-03 08:50:33,486: Loading model XML from file /home/wzy/src/Firmware/Tools/sitl_gazebo/models/iris/iris.sdf
[rosout][INFO] 2020-03-03 08:50:33,491: Waiting for service /gazebo/spawn_sdf_model
[rosout][INFO] 2020-03-03 08:50:33,493: Calling service /gazebo/spawn_sdf_model
[rospy.internal][INFO] 2020-03-03 08:50:33,496: topic[/clock] adding connection to [http://wzy-lab:40509/], count 0
[rospy.internal][INFO] 2020-03-03 08:50:33,769: topic[/rosout] adding connection to [/rosout], count 0
[rosout][INFO] 2020-03-03 08:50:33,774: Spawn status: SpawnModel: Successfully spawned entity
[rospy.core][INFO] 2020-03-03 08:50:33,775: signal_shutdown [atexit]
[rospy.internal][INFO] 2020-03-03 08:50:33,776: topic[/rosout] removing connection to /rosout
[rospy.internal][INFO] 2020-03-03 08:50:33,776: topic[/clock] removing connection to http://wzy-lab:40509/
[rospy.impl.masterslave][INFO] 2020-03-03 08:50:33,777: atexit
```

julianoes, 2020-03-03:
Sorry, it's not about the dependencies of `px4` but `gzserver`, however,  they are loaded dynamically I believe, so `ldd` might not show it.

When you do `make clean`, it should not say:
```
[0/4] Performing build step for 'sitl_gazebo'
ninja: no work to do.
```
but actually rebuild Gazebo, does it do that?

julianoes, 2020-05-25:
No answer, closing.

closed 2020-05-25 by julianoes

fabrizioschiano, 2020-08-31:
Hi @julianoes , I have the same problem when I run `make px4_sitl_default gazebo `. Could you help with that? @myboyhood , how did you solve the problem, if you did? 

Thanks

fabrizioschiano, 2020-09-01:
I solved my problem I think by doing both `make clean` in the `Firmware` folder and `rm -r ~/catkin_ws/devel ~/catkin_ws/build` (to clean the ROS catkin workspace).

I think that some library paths (e.g., `LD_LIBRARY_PATH`) got corrupted but I don't know for sure. 
Now the `make px4_sitl_default gazebo` works and I am able to see the iris drone in the gazebo environment and I can also give commands such as `commander takeoff` and `commander land`. 

I hope this will help someone else in the same situation

myboyhood, 2020-09-03:
> I solved my problem I think by doing both `make clean` in the `Firmware` folder and `rm -r ~/catkin_ws/devel ~/catkin_ws/build` (to clean the ROS catkin workspace).
>
> I think that some library paths [...]

thanks,I will try your method!!

HuaYuXiao, 2024-12-07:
> thanks,I will try your method!!

Yes, you are definitely right! I encoutered the same problem, and finally solved it with this method!

HuaYuXiao, 2024-12-07:
> I solved my problem I think by doing both `make clean` in the `Firmware` folder and `rm -r ~/catkin_ws/devel ~/catkin_ws/build` (to clean the ROS catkin workspace).
>
> I think that some library paths [...]

Yes, this is definitely right! I encoutered the same problem, and finally solved it with this method!