PX4/PX4-Autopilot #14165: EKF2 estimates for x and y unstable with constant optical flow innovation

dominiknatter, 2020-02-15:
Hey everyone,
first of all shoutout for your great work, it's really impressive.

Remark: In hindsight I've written quite a long story here, but I hope providing what I figured out help digging into the problem faster.

I'm currently working with the Intel Aero RTF drone with the PX4 v1.9.0 firmware. I'm trying to fly indoors (hence no GPS) with ORB-SLAM2. This basically works already, fusing the data to the EKF2 via /mavros/vision_pose/pose. However, I experienced undesired behavior if SLAM fails (e.g. when it looses track after fast motions). I thought it should be possible to still get decent state estimates using only internal sensors, PX4Flow, and the LeddarOne distance sensor (the sonar on the PX4Flow board is still active but not used. Also I'm sharing the Telemetry port for PX4flow and LeddarOne as suggested here https://github.com/intel-aero/meta-intel-aero/issues/315#issuecomment-391439235).

So I did some experiments without SLAM running, and dragging around the drone by hand only for safety. Indeed, the position and velocitiy estimates are quite ok for multiple minutes (note that I put down the drone from time to time to observe the integration drifts). But at some point x and y drift off devastatingly and need to be resetted by the EKF2 all the time. I'm not sure why this happens. All I found out (using fw_px4_plottools) is that the optical flow innovations just freeze at some point and that's exactly the timestamp when everything gets messy. I'm not sure whether that's the cause or just another effect though. Note that also lifting the drone to a height where optical flow values should be plausible didn't make it recover anymore.

The z position / velocity do not seem to be affected. However, I noticed an issue there as well, although I personally can't see any connection: there are non-zero z-velocities (up to 0.4m/s), even when the drone is standing still and the z position is constant. Thinking of their kinematic relationship, that doesn't really make sense. Some parts of the plots look like they behave the same way, i.e. z velocity is just a scaled version of z position (which would explain why it's static and then it'd only be a bug in the log). Also looking at different static phases, the z positions are not necessarily equal (although the floor is the same). I saw that the LeddarOne tends to output 0 when the drone is on the floor, so I tried to fix this by forcing EKF2_MIN_RNG to 0.0m, but it didn't do the trick.

Here are two logs where the drone is dragged around and xy goes nuts after some minutes.
https://logs.px4.io/plot_app?log=91bc8ed4-6044-45e8-bfcf-5416976ad011
https://logs.px4.io/plot_app?log=907f8a65-388e-40ad-83d9-3bdee61bc29c

Here's one log where I actually flew around with the RC using POSITION flight mode. Note how here first OF innovations go constant and simultaneoulsy xy drift off. But then when SLAM is fed xy recovers. However, apparently I lost SLAM at some point during flight and the drone skyrocketed to the left (as far as I can tell at a faster rate than I thought MPC_XY_VEL_MAX would allow).
https://logs.px4.io/plot_app?log=af47b24a-1a44-4eee-898d-7f9a72c8412b

Here are also 2 screenshots to show the main problem without the need of opening any logs, i.e. drifting xy position (and velocity) and constant optical flow innovations:
[image]

The third case makes clear why I want to make sure, a scenario without SLAM would lead to a safe landing at least. Hopefully some more experienced people can have a look at this and we find out whether it's my setup, some parameters, or the EKF2 (for the latter I assume @priseborough to be the ideal contact person). I'm of course happy to provide more info on my setup if needed.

Best,
Dominik

mcsauder, 2020-02-16:
Hi @dominiknatter , I'm not an expert by any stretch in this domain, but I think what you are seeing are the effects of observability (in the controls engineering use of the term).  Basically, if you can set up a series of state space equations, what you will find is that without the SLAM solution, the relative position/velocity X/Y measurements and relative Z (distance sensor) measurements do not provide a data source with bounded error propagation.  In short, the reason it works for a while is because the errors propagating are relatively small for a time period, but as time passes their uncertainty grows.  Back to observability, without an observation of the actual state with bounded error, (such as GPS or SLAM observation/measurement), the rank of your state space matrices will not be equal to the number of states you are tracking and the solution eventually blows up due to the unbounded uncertainty.  If you can create a solution to this problem you will become a very famous mathematician! :)

I hope this makes some sense.  And, to anyone who knows the subject better than I do, please feel free to correct anything I might be mistaken about!

-Mark

dominiknatter, 2020-02-16:
Hello @mcsauder,

thanks for your reply. I see your concerns about the system's observability and, although haven't analyzed them mathematically, I agree with you - pure integration is not enough. However, I'll try to explain why I think it's not the problem here:
 - as you mentioned, the errors would grow over time. That's reasonable and true as you can see in the following plot. I plot the variance of the x position exemplarily for the same flight as plotted above. When optical flow innovations are still ok, the variance grows at a reasonable pace, just as we would expect from pure integration. However, after optical flow fails the growth rates are multiple magnitudes larger (that's why I had to plot the same graph zoomed in, to show how small they are in the beginning).
[image]

 - what further doesn't fit your explanation: in the one log I provided the unbounded growth happens quite early (<2 minutes in). Also in that bag, optical flow innovations fail early. I don't think that's a coincidence.

 - finally, many people can fly or at least hover with this set of sensors, as many videos suggest (e.g. https://www.youtube.com/watch?v=MtmWYCEEmS8)

So to summarize: I agree, endless flight is probably not possible without any absolute measurements for x and y position due to increasing errors. However, I want to achieve reasonably growing uncertainty and that's clearly not the case here. And my assumption is because optical flow is not incorporated well at some point.

BTW I realized I can't find optical flow messages in the log. I wanted to see whether they just stop being streamed at some point (so it would be a connection issue or a faulty PX4flow board). Luckily, for the third log I recorded a ROS bag in parallel. It incorporated /mavros/px4flow/raw/optical_flow_rad and that topic still offers varying values even after the crash, although OF innovations froze over a minute earlier.

Best,
Dominik

LorenzMeier, 2020-02-16:
The video you are referencing actually proves the point from @mcsauder: In altitude hold mode the pilot is in the loop (external feedback loop), in position hold mode there is optical flow.

Inertial sensors allow you to bridge the gap as well.

dominiknatter, 2020-02-16:
Thanks for chiming in @LorenzMeier! Oh I thought @mcsauder is trying to say optical flow is not enough to close the loop ("...without an observation of the actual state with bounded error, (such as GPS or SLAM observation/measurement)..."). I agree that optical flow is needed and am not expecting reasonable results with only inertial sensors. Still, I don't know why optical flow stops working at some point.

Maybe for clarification the issue should be renamed to "optical flow innovations freeze" or similar. I didn't do that so far as I was not sure whether that's the problem for the destabilization, or whether it's just another effect of something I haven't noticed yet.

dominiknatter, 2020-02-17:
The missing optical flow values are my bad, I didn't log them. So I fixed that and recorded three more logs. All have in common that a spike in the OPTICAL_FLOW_RAD.integration_time_us field indicates the start of EKF2 drifts (i.e. optical flow values do not seem to be incorporated anymore). Often, but not always, this spike comes along with a spike in the sonar distance value to 65m (I guess that's just the maximum value uint16_t can hold).
An example is seen here: In the first image we see that optical flow innovations freeze and positions start to drift off wildly, in the second image I listed some optical flow data (note the log scales).
[image]
[image]

Now that I narrowed it down I wonder
a) whether it's exactly this problem and was discovered over a year ago already:
https://github.com/PX4/Flow/issues/102
b) whether it's due to a PX4Flow internal problem or due to my setup, i.e. using I2C for communication. I could imagine disconnecting GPS and using that (UART) port for the PX4Flow instead.

Any thoughts on this?

dominiknatter, 2020-03-19:
btw we didn't figure out what caused the spikes in the optical flow messages. However, it seems that these spikes make the EKF2 stop using optical flow at all (hence the afterwards constant innovations I guess), so we stopped the EKF2 from incorporating faulty data. With a quite hacky workaround we just ignore messages with too high integration timespans (we determined 1 second as a threshold) here
https://github.com/PX4/Firmware/blob/v1.9.0/src/modules/ekf2/ekf2_main.cpp#L1162

bresch, 2020-03-20:
@dominiknatter I think we could add that logic to ignore the samples with too large integration time span. Could you open a PR to PX4/Firmware with the modification that works for you please?

victordias2000, 2020-05-24:
Hi @dominiknatter, we're in a very similar scenario as yours (Intel Aero RTF, PX4Flow, flying indoors, and getting X and Y start growing off the mark). Two questions related to this: 

1-) Did your workaround stabilise X and Y values
2-) Do you mind explaining or sharing your workaround?

Thanks in advance,
Victor.

dominiknatter, 2020-05-25:
@victordias2000
1) first of all, in the tests we conducted for this problem we simply dragged the Intel Aero around - it didn't fly on its own. In that scenario, our fix removed the immense drifts that occured after a certain point. We never tested stabilization in the air separately, but I assume with properly calibrated internal sensors and well-tuned optical flow drift should be slow or non-existent.

2) As far as I remember, we simply added another check [here](https://github.com/PX4/Firmware/blob/v1.9.0/src/modules/ekf2/ekf2_main.cpp#L1164) like `if flow.dt < 1000000`, so we only accepted integration times below 1 second. (Pay attention, in newer firmware versions flow.dt is transformed to seconds earlier, see e.g. [here](https://github.com/PX4/Firmware/blob/master/src/modules/ekf2/ekf2_main.cpp#L1060)). However, I was not able to access our hardware since my last post due to COVID-19, so I cannot give you the exact code. @bresch this is also why I didn't come up with a pull request yet, as I wanted to be sure I add the correct code. I am sorry for that and hope I can go there in June.

dominiknatter, 2020-06-29:
Hi @bresch. Finally made it to the office. It seems like the proposed fix was just as trivial as expected. I just wondered whether the hard-coded threshold of 1 second is ok, but I assume most sensors work at faster rates. Let me know if I can be of any further help

bresch, 2020-06-29:
>I just wondered whether the hard-coded threshold of 1 second is ok

Most of the other timeouts are also hardcoded, so I think it's fine like that. I we see that we need to have it different on some setups, we can still define a parameter later.

PR #15237 (2020-06-29): Ignore optical flow samples with too large integration time spans.
Fixes #14165. Probably also fixes PX4/Flow#102

Within ekf2, optical flow messages (amongst others) are fused to the state estimates. It might occur that optical flow sensors report unreliable and unrealistic spikes. In that case, the state estimator went crazy so far and just ignored optical flow values from that moment on.

The common thread for all these spikes seems to be a too high integration time span (can be as high as the data type upper limit). Therefore, this fix adds a simple logic that ignores unrealistically high  [... cut]
commit d8831c51 (2020-06-29): ekf2: Ignore optical flow samples with too large integration time spans. Fixes #14165

Within ekf2, optical flow messages (amongst others) are fused to the state estimates. It might occur that optical flow sensors report unreliable and unrealistic spikes. In that case, the state estimator went crazy so far and just ignored optical flow values from that moment on.

The common thread for all these spikes seems to be a too high integration time span. Therefore, this fix adds a simple logic that ignores unrealistically high integration time spans. As a threshold, 1 second was chosen.

Reported-by: [... cut]
closed 2020-06-29 by dagar