PX4/PX4-Autopilot #14169: Add support for multi-part STATUSMESSAGE

hamishwillee, 2020-02-17:
Add support for chunked [STATUSTEXT](https://mavlink.io/en/messages/common.html#STATUSTEXT( messages. Info about change here: https://github.com/mavlink/mavlink/pull/1295

dayjaby, 2020-03-03:
I can give it a try. Short question: why does it say "Status text message, without null termination character" in https://mavlink.io/en/messages/common.html#STATUSTEXT? The null termination character looks necessary as we have no field containing the length of the string and it is used in the current px4 statustext implementation.

julianoes, 2020-03-03:
The spec means that you can use all 50 bytes. If you use all 50 bytes, then the other side should only use these 50 bytes and then terminate, and not read over that limit.

E.g.:
https://github.com/mavlink/MAVSDK/blob/00961b9813f810c1bc89d372680f8e099ad056b2/src/plugins/telemetry/telemetry_impl.cpp#L809-L812

dayjaby, 2020-03-04:
So this would be correct behaviour?

```
STATUSTEXT {severity : 6, text : [logger] file: ./log/2020-03-04/06_12_52.ulg, id : 0, chunk_seq : 0}
STATUSTEXT {severity : 6, text : ARMED by Arm/Disarm component command, id : 1, chunk_seq : 0}
STATUSTEXT {severity : 6, text : Using minimum takeoff altitude: 2.50 m, id : 2, chunk_seq : 0}
STATUSTEXT {severity : 6, text : Takeoff detected, id : 3, chunk_seq : 0}
STATUSTEXT {severity : 6, text : Landing at current position, id : 4, chunk_seq : 0}
STATUSTEXT {severity : 6, text : Landing detected, id : 5, chunk_seq : 0}
STATUSTEXT {severity : 6, text : DISARMED by Auto disarm initiated, id : 6, chunk_seq : 0}
STATUSTEXT {severity : 6, text : ARMED by Arm/Disarm component command, id : 7, chunk_seq : 0}
STATUSTEXT {severity : 6, text : [logger] file: ./log/2020-03-04/06_02_49.ulg, id : 8, chunk_seq : 0}
STATUSTEXT {severity : 6, text : Using minimum takeoff altitude: 2.50 m, id : 9, chunk_seq : 0}
STATUSTEXT {severity : 6, text : Takeoff detected. Test a very long string. ABCDEFG, id : 10, chunk_seq : 0}
STATUSTEXT {severity : 6, text : HIJKLMNOPQRSTUVWXYZ, id : 10, chunk_seq : 1}
```

In the last line I logged "Takeoff detected. Test a very long string. ABCDEFGHIJKLMNOPQRSTUVWXYZ". If yes, I gonna create a PR for this.

julianoes, 2020-03-04:
In my opinion that is correct. Thanks.

julianoes, 2020-06-02:
Implemented in #14295.

closed 2020-06-02 by julianoes