PX4/PX4-Autopilot #14171: Precision Landing from Offboard

felix-west, 2020-02-17:
As the title says, I'm looking for a way to initiate precision landing from offboard mode. I have a camera connected to my companion computer which gives the position and attitude of the target to the offboard computer, and that redirects the message to the flight controller. Currently messages of the type "LANDING_TARGET" ( Mavlink #149 ) are being used, so the flight controller can read and use the messages. 

The discussion at https://github.com/PX4/Firmware/pull/9050 and the subsequent video of the precision landing at https://www.youtube.com/watch?v=HGX1RccfN1g both seem to use mission mode to initiate the precise landing. In my case that is not easily doable, I basically want to send a command that starts precise landing. The discussion at https://discuss.px4.io/t/visual-precision-landing/2418 suggests doing exactly that, except it is not actually stated HOW or if it actually worked. In the discussions and in the code I saw that precision landing is its own mode, such as offboard or mission modes. Changing from offboard to precision landing mode would be a feasible work-around for me. However, looking through the documentation, I don't see how the modes are actually changed. As in, which Mavlink message is sent with which parameters. I know this must be possible because that is exactly what my companion computer does right now using MAVSDK. However, the same case applies here: looking through the documentation does not show what exactly is sent to change modes. 

The next issue is that while trying to force my drone to start a precision landing, I changed the firmware code of the drone to reflect that. I changed most occurrences of "VEHICLE_COMMAND_VEHICLE_CMD_NAV_LAND" to "VEHICLE_COMMAND_VEHICLE_CMD_NAV_PRECLAND" and "VEHICLE_CMD_NAV_LAND" to "VEHICLE_CMD_NAV_PRECLAND" hoping to force it to go into precision landing mode. However, it changed nothing. From the debug messages and the logging I can clearly see that the drone still goes into "VEHICLE_CMD_NAV_LAND" or normal landing. Because I do not use global coordinates, only local coordinates using camera and UWB, I disabled "status_flags.condition_global_position_valid" in src/commander/state_machine_helper.cpp, but it changes nothing. The only thing I have yet to see is if the messages from the camera/companion computer are being accepted by the flight controller. However, according to the companion the messages are at least valid mavlink messages. 

I also saw another discussion to exactly this topic a while back, asking for precise landing in offboard mode. However, I can't seem to find it, and the last status of that discussion was that it might be worked on in the future.

julianoes, 2020-03-02:
@okalachev maybe you can help here?

okalachev, 2020-03-02:
Hi, you can easily change the mode to precision landing using `MAV_CMD_DO_SET_MODE` command.

The main mode is `MAV_MODE_FLAG_SAFETY_ARMED | MAV_MODE_FLAG_CUSTOM_MODE_ENABLED`, the custom mode is `PX4_CUSTOM_MAIN_MODE_AUTO` (4), the custom sub mode is `PX4_CUSTOM_SUB_MODE_AUTO_PRECLAND` (9).

felix-west, 2020-03-17:
Ok so after a month I almost have to give up on this project because it seems my use case is not supported at all. Since I only have local coordinates to work with, precision landing does not work correctly. First of all, without changing the firmware code it does not go into this mode at all. To go into this despite that, I commented out the requirement for valid GPS coordinates in the state machine helper cpp. Because the precision landing works on the auto mode position controller, it apparently only uses GPS coordinates for position control and the local NED coordinates (note: the coordinates have to be in NED frame and not body frame) for more precise control of the latitude and longitude values used for positioning. To generate these coordinates I used the testing mode of the GPS module, but instead of using constant values as output I calculate latitude and longitude in the GPS module by using my local NED coordinates. This way the precision landing controller should be able to work either way. 

However, when testing I found two major issues with the precision landing module using this method. First, having a negative value for altitude immediately stops all motors, even if the drone is actually still in the air (even if it has vision of a target). This is also true for normal landing. Since my targets have differing heights, I have to account for negative altitude values, and this does not work. Secondly, this method does extremely weird things with the position setpoints. I noticed that when the landing procedure starts the value of the setpoint using local coordinates cannot be changed anymore. Even when sending landing target messages via camera and onboard controller I cannot influence the value of pos_sp anymore at all. I've tried using mavlink messages #84 and #149 and neither of them does anything. Depending on what I comment out in the precland code, things get even weirder. It can happen that my old setpoint gets carried over to the landing procedure, but then never changes again afterwards, even if the drone is in horizontal or vertical approach and the setpoints should be changing every time. It can also happen that some other point gets chosen between the old setpoint and 0, and then progressively gets smaller. Again, nothing I do changes what these setpoints become. 

As far as I am concerned, I can't get precision landing without GPS to work. For me the only way to do this now seems to be through my companion stm32. Luckily I was already able to make MAVSDK and my stm communicate. Now I will try to make the stm control the drone through offboard messages instead of MAVSDK, and simply use that on my PC to give the coordinate information to the stm. By giving a series of setpoints using the UWB and cameras connected to the microcontroller I should be able to approach the target similar as in precision landing, and issue a kill or land command once I think I landed or lose connection to the PC. It seems like any other method using the firmware simply makes no progress at all.

felix-west, 2020-03-19:
A quick update, I finally found the culprit of my issues. It was in fact a the function in the precision landing horizontal approach state for calculating the new setpoint position. When removing "map_projection_reproject(...)" and replacing that with my own function to calculate new setpoints it seems to work now.

closed 2020-03-19 by felix-west