PX4/PX4-Autopilot #14185: ERROR while building

islamdib, 2020-02-19:
Hello,
I have pixhawk v2 
after Downloading PX4 Source Code I tried to build for Pixhawk board.
An error was occured.

dib@dib-WM15-CCLPRO:~/Firmware$ make px4_fmu-v2_default
-- PX4 version: v1.10.0-beta4-469-gfbcc6a96bc
-- PX4 config file: /home/dib/Firmware/boards/px4/sitl/default.cmake
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- cmake build type: RelWithDebInfo
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-107-g1813034
-- Configuring done
-- Generating done
-- Build files have been written to: /home/dib/Firmware
Error: could not load cache
Makefile:193: recipe for target 'px4_fmu-v2_default' failed
make: *** [px4_fmu-v2_default] Error 1

What the error source is ?

julianoes, 2020-03-02:
Have you tried `make clean` ?

I found this: https://stackoverflow.com/questions/16319292/cmake-error-could-not-load-cache

julianoes, 2020-06-02:
Closing, there was no response.

closed 2020-06-02 by julianoes