PX4/PX4-Autopilot #14189: SITL gazebo real time factor is slowing down for plane

dzambrano, 2020-02-19:
**Describe the bug**
The real time factor for a simple plane simulation progressively slows down. From about 16 to 5 in 10 minutes. Similar to [#12975](https://github.com/PX4/Firmware/issues/12975)

**To Reproduce**
Firmware version 10.0
Gazebo 9.0

```
git checkout v1.10.0
git submodule update --init --recursive
pip install jinja2 empy catkin_pkg toml numpy
export PX4_SIM_SPEED_FACTOR=20
make px4_sitl gazebo_plane
```
Then in the px4 console:
```
pxh> commander takeoff
```

**Expected behavior**
The real time factor goes down consistently. 

**Log Files and Screenshots**
*Always* provide a link to the flight log file:
Log files and gz stats are available to the following link:
[Log files](https://drive.google.com/drive/folders/1aN8FYOeleXipdWKhOEPBMHR30sqFBS6t?usp=sharing)

julianoes, 2020-03-02:
Thanks for the issue. I'll put this on my list to reproduce.

shrit, 2020-03-03:
I think this is related to SDF files, which has been updated in last year. It slows down since we have added magnetometers and gyro meters parameters in the SDF. The last commit I remember working without a slowing down in this one 0e43ddafa0c70a190a9080c17c9540d8f7ab5261

vwueest, 2020-03-17:
Having the same problem. @shrit  I tried to checkout the commit you mentioned, but I get 
`fatal: reference is not a tree: 0e43ddafa0c70a190a9080c17c9540d8f7ab5261`
I also compared the sitl_gazebo from PX4 1.9.2 (which was released before the commit you sent) with the current one, but the `plane.sdf` file is the same.
Do you have a suggestion on work around or how to fix it? The simulation slow down is blocking us quite a bit sadly

shrit, 2020-03-17:
@vwueest I do not know why you have this error, but you need to checkout this commit from inside the submodules (you have to go inside sitl_gazebo and checkout the commit), not in the PX4 firmware. 
After that, you commit the detached head and create your own branch to save this commit.
Finally, at the PX4 git level, you create a new branch, add the new submodules commit and then recompile the entire PX4 firmware.
If you have the appropriate PX4 version, you should not have any problem during the compilation.

shrit, 2020-03-17:
Relevant ticket: https://github.com/PX4/Firmware/issues/12975
A closely related issue still open :+1: 
@julianoes if you have an idea where I can start debugging I can work on a pull request.

vwueest, 2020-03-30:
@shrit thanks for your answer. I tried checking out the old commit as you suggested. The problem is that we need px4>=v1.10.0, because we control the plane offboard and that was not working properly for fixed wings before v1.10.0. 
When i checkout Gazebo_SITL it is not compatible with any of those px4 versions. @julianoes, @shrit do you know if this issue is still being looked at? We would highly appreciate some advances on this front. Thanks a lot

shrit, 2020-03-30:
@vwueest You are welcome.
Actually, I do not have a clear idea of why this issue is happening in the first place. I am also concerned with this issue since it is affecting quadrotors. Let us see what @julianoes thinks about it.
The issue is more clear if you run long simulations for about 5 to 10 hours, or for several days. 
I have also noticed the same thing in this video, it should be similar to what is happening with your simulation I think (Check out the last 10 seconds): https://youtu.be/aEzFKPMEfjc

julianoes, 2020-03-31:
I'm currently not looking at that. I want to but it's not as urgent as everything else :confused:, at least for me.

And I haven't done that sort of debugging before. I would start with doing some profiling. First profile it for 1 minute and safe the callgrind data. Then profile it for 15 mins and save the callgrind data and then compare it and check what increased.

closed 2020-05-14 by julianoes