PX4/PX4-Autopilot #14195: Thrust drops out on multicopter in altitude-controlled flight modes (FMUv3, PX4 v1.10.1)

Finwood, 2020-02-20:
When flying in altitude-controlled flight modes (Altitude Mode, Position Mode), a hexacopter running PX4 v1.10.1 on an FMUv3 (the [Cube]) repeatedly looses all thrust for a short while, resulting in a kind of "hiccup": [thrust setpoint, actuator outputs]

[[image]][log]
[[image]][log]

This seems to happen mainly when the copter is not moving horizontally: [thrust setpoint, velocity]

[[image]][log]
[[image]][log]

The corresponding log file can be found [here][log].

A few random remarks:
- prior to the update to v1.10 (running v1.9.2), the drone flew stable
- hexacopter, around 3 kg takeoff mass, fairly overpowered
- I described aspects of this problem [on fhe forum][forum]
- before lowering [`IMU_GYRO_RATEMAX`] from 400 Hz to 250 Hz, I got warning messages of I2C1 being low on stack ([log][other log], see [forum post][forum] for details)

  [log]: https://logs.px4.io/plot_app?log=1c36e82a-523f-4c10-ad05-06028aee0b4e
  [cube]: https://docs.px4.io/v1.10/en/flight_controller/pixhawk-2.html
  [forum]: https://discuss.px4.io/t/thrust-dropouts-in-altitude-mode-after-update-to-v1-10/15259
  [`IMU_GYRO_RATEMAX`]: https://docs.px4.io/v1.10/en/advanced_config/parameter_reference.html#IMU_GYRO_RATEMAX
  [other log]: https://logs.px4.io/plot_app?log=e748a742-a3a7-40d7-bcde-c5bcd7b46ed9

bkueng, 2020-02-21:
Hi
There are in-air landing detections and you need to set the hover throttle (MPC_THR_HOVER) to around 20%.

Finwood, 2020-02-21:
Seems legit :man_facepalming: 

MPC_THR_HOVER and MPC_THR_MIN were configured properly before updating to v1.10, most likely something went wrong importing the v1.9 parameters.

I will test later today to confirm :+1:

Finwood, 2020-02-21:
> you need to set the hover throttle (MPC_THR_HOVER) to around 20%

Yep, that fixed it. Thanks! :+1:

closed 2020-02-21 by Finwood