PX4/PX4-Autopilot #14201: Beaglebone Blue MPU9250 Error

Superbear88, 2020-02-21:
The MPU9250 get initialiced and then, the mag not working

ERROR [mpu9250] I2C transfer to mag timed out

See Output

INFO  [px4] assuming working directory is rootfs, no symlinks needed.

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh /home/debian/px4/px4.config 0
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
BMP280_I2C on I2C bus 2 at 0x76
MPU9250_I2C on I2C bus 2 at 0x68
AK8963_I2C on I2C bus 2 at 0x0c

PX4 Exiting...
ERROR [mpu9250] I2C transfer to mag timed out
ERROR [mpu9250] I2C transfer to mag timed out
ERROR [mpu9250] I2C transfer to mag timed out
ERROR [mpu9250] I2C transfer to mag timed out
INFO  [commander] LED: open /dev/led0 failed (22)
ERROR [sensors] Accel #0 fail:  STALE!
ERROR [sensors] Sensor Accel #0 failed. Reconfiguring sensor priorities.
WARN  [sensors] Remaining sensors after failover event 0: Accel #0 priority: 1
ERROR [sensors] Gyro #0 fail:  STALE!
ERROR [sensors] Sensor Gyro #0 failed. Reconfiguring sensor priorities.
WARN  [sensors] Remaining sensors after failover event 0: Gyro #0 priority: 1
INFO  [mavlink] mode: Normal, data rate: 1000000 B/s on udp port 14556 remote port 14550
INFO  [linux_pwm_out] Loaded mixer from file ROMFS/px4fmu_common/mixers/quad_x.main.mix
INFO  [linux_pwm_out] Starting PWM output in bbblue_rc mode
INFO  [drivers_board] Initializing librobotcontrol ...
INFO  [logger] logger started (mode=all)
INFO  [mavlink] using network interface wlan0, IP: 192.168.1.131
INFO  [mavlink] with netmask: 255.255.255.0
INFO  [mavlink] and broadcast IP: 192.168.1.255
INFO  [mavlink] partner IP: 192.168.1.195
INFO  [px4] Startup script returned successfully

Gyro works in GC

SalimTerryLi, 2020-02-21:
Which version did you try? This happens just after the purge of DF but now it is not happening anymore on master. I'm using RPi, which has almost the same condition with BBB.

Superbear88, 2020-02-21:
Hi 

HW arch: BEAGLEBONE_BLUE
FW git-hash: 86bf4b2289c1630833c920c84dfc95a82e2ac6b5
FW version: 1.11.0 0 (17498112)
FW git-branch: master
OS: Linux
OS version: Release 4.14.71 (68044799)
Build datetime: Feb 18 2020 12:30:49
Build uri: localhost
Toolchain: GNU GCC, 6.3.1 20170404
PX4GUID: 10084c424242444945553030303030303030
UNKNOWN MCU

SalimTerryLi, 2020-02-21:
I doubt it is caused by https://github.com/PX4/Firmware/blob/86bf4b2289c1630833c920c84dfc95a82e2ac6b5/src/drivers/imu/mpu9250/mpu9250.h#L46-L48

Superbear88, 2020-02-21:
Do you have a solution for it?
librobotcontrol can controll it without problems... 

Thank you

dagar, 2020-02-21:
The MPU9250 on I2C has been neglected and needs a closer look.

SalimTerryLi, 2020-02-21:
> The MPU9250 on I2C has been neglected and needs a closer look.

I found this maybe the cause https://github.com/PX4/Firmware/blob/86bf4b2289c1630833c920c84dfc95a82e2ac6b5/src/drivers/imu/mpu9250/MPU9250_mag.cpp#L229-L233
It should be wrappered as write_reg. My fault.