PX4/PX4-Autopilot #14204: EKF2 performance regression

dagar, 2020-02-21:
EKF2 cpu usage has increased considerably.

PX4/Firmware ecl update timeline: https://github.com/PX4/Firmware/commits/master/src/lib/ecl

### Timeline

 - 2019-09-26: `EKF2 update: 41151 events, 17355386us elapsed, 421.75us avg, min 19us max 2978us 708.544us rms` 
 - ?? 2020-01-20: Jan 9 ecl update https://github.com/PX4/Firmware/commit/4e8665082ea785d5b2c7c485824658f694319150#diff-eab0fded47ba453da12bc405ab7b6d5f
 - 
 - 2020-02-21: `ekf2: update: 11127 events, 7935146us elapsed, 713.14us avg, min 18us max 2033us 874.154us rms`

bresch, 2020-03-03:
Tested on Pixhawk4 (fmuv5), nothing attached, disarmed

> Mar 02, 2020
3be8b3d177
ekf2: update: 20247 events, 7280866us elapsed, 359.60us avg, min 12us max 979us 401.864us rms

> Feb 12, 2020
17ab617877
ekf2: update: 20942 events, 7660620us elapsed, 365.80us avg, min 11us max 1001us 410.382us rms

> Feb 05, 2020
1aa514a8ea
ekf2: update: 2940 events, 1075008us elapsed, 365.65us avg, min 12us max 949us 1301.413us rms

> Jan 08, 2020
92adc5afaa
ekf2: update: 11847 events, 4322963us elapsed, 364.90us avg, min 12us max 978us 634.418us rms

> Dec 05, 2019
d7b95870b9
ekf2: update: 61246 events, 17271477us elapsed, 282.00us avg, min 12us max 805us 339.088us rms

> Nov 27, 2019
09c8c8f706
ekf2: update: 140716 events, 26891873us elapsed, 191.11us avg, min 12us max 828us 217.603us rms

> Nov 25, 2019
63e99db29b
ekf2: update: 73300 events, 14153166us elapsed, 193.09us avg, min 12us max 812us 212.825us rms

> Nov 02, 2019
f0ac270174
ekf2: update: 59500 events, 11424500us elapsed, 192.01us avg, min 11us max 954us 222.121us rms

> Oct 27, 2019
17b654f165
ekf2: update: 8677 events, 1719439us elapsed, 198.16us avg, min 12us max 947us 238.719us rms

> Oct 03, 2019
5f040fe24
ekf2: update: 21027 events, 4149392us elapsed, 197.34us avg, min 13us max 934us 260.660us rms
ekf2: interval: 21027 events, 4007.11us avg, min 3699us max 4984us 93.594us rms
ekf2: cycle time: 21027 events, 6405696us elapsed, 304.64us avg, min 56us max 1127us 320.322us rmsc

> Sep 26, 2019
190c817a9e
EKF2 update: 12079 events, 2674740us elapsed, 221.44us avg, min 12us max 1118us 275.075us rms
EKF2 data acquisition: 12079 events, 784037us elapsed, 64.91us avg, min 37us max 351us 63.158us rms

bresch, 2020-03-03:
ecl diff between 09c8c8f and d7b9587:
https://github.com/PX4/ecl/compare/6b5f011bc224a45fd57d33e90cc7a136aa075b6c...181303488cfc4d2ae34f61f0ad9a34b49a2501f7

closed 2020-03-04 by bresch